OctoPrint/OctoPrint #4856: Virtual Printer Error: Missing checksum

GordonW-Printer, 2023-07-20:
### The problem

My configuration is a Raspberry Pi Zero 2W with the latest stable OctoPi/OctoPrint image dated 20030718.
I have my Upload folder SMB mouted from a Synology NAS and have not noticed any issues with this configuration.
Real printer is Prusa MK3S+ with the Raspberry Pi integrated.
I have been using the same confiuguration for 3 years with the exception of upgrading from Raspberry Pi Zero to 2W model and multiple OctoPi/OctoPrint versions.

While using the "VIrtual Printer" to investigate another (unrelated issue) I observed the following errors reported that are not seen on the physical printer using the exact same gcode file.

2023-07-20 14:50:21,186 >>> ok
2023-07-20 14:50:21,189 <<< N99 G1 X113.286 Y104.366 E.03978*110

2023-07-20 14:50:21,190 >>> ok
2023-07-20 14:50:21,193 <<< N100 G1 X113.474 Y102.813 E.04905*86

2023-07-20 14:50:21,194 >>> Error: Missing checksum
2023-07-20 14:50:21,195 >>> Resend:100
2023-07-20 14:50:21,199 >>> ok
2023-07-20 14:50:21,202 <<< N100 G1 X113.474 Y102.813 E.04905*86

2023-07-20 14:50:21,205 >>> ok
2023-07-20 14:50:21,208 <<< N101 G1 X113.679 Y101.898 E.02938*80

2023-07-20 14:50:21,210 >>> ok
2023-07-20 14:50:21,214 <<< N102 G1 X114.469 Y99.815 E.06987*98

***
Skipping some lines
***

023-07-20 14:50:21,224 >>> ok
2023-07-20 14:50:21,226 <<< N105 G1 X116.288 Y97.71 F10800*126

2023-07-20 14:50:21,228 >>> Error: Missing checksum
2023-07-20 14:50:21,229 >>> Resend:105
2023-07-20 14:50:21,233 >>> ok
2023-07-20 14:50:21,236 <<< N105 G1 X116.288 Y97.71 F10800*126

***
Skipping some lines
***

2023-07-20 14:50:22,258 >>> ok
2023-07-20 14:50:22,261 <<< N108 G1 X118.074 Y96.002 E.04905*101

2023-07-20 14:50:22,263 >>> ok
2023-07-20 14:50:22,266 <<< N109 G1 X118.797 Y95.481 E.02795*99

2023-07-20 14:50:22,268 >>> ok
2023-07-20 14:50:22,272 <<< N110 G1 X119.736 Y94.97 E.03351*94

2023-07-20 14:50:22,273 >>> Error: No Line Number with checksum, Last Line: 109
2023-07-20 14:50:22,277 <<< N111 G1 X120.69 Y94.488 E.03351*82

2023-07-20 14:50:22,279 >>> Error: expected line 110 got 111
2023-07-20 14:50:22,280 >>> Resend:110
2023-07-20 14:50:22,285 >>> ok
2023-07-20 14:50:22,289 <<< N110 G1 X119.736 Y94.97 E.03351*94

2023-07-20 14:50:22,291 >>> ok
2023-07-20 14:50:22,294 <<< N111 G1 X120.69 Y94.488 E.03351*82

Created new OctoPi/OctoPrint image on a Raspberry Pi 3B+ to verify it was not the 2W hardware causing the issue and I see the same errors in the Virtual printer log, I didn't connect this to a physical printer as I was just validating the Pi H/W.

 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.2

### Operating system running OctoPrint

OctoPi 1.0.0

### Printer model & used firmware incl. version

Virtual Printer Plugin

### Browser and version of browser, operating system running browser

Fedora-F38, Firefox 115.0.2 (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

System bundle from Raspberry Pi Zero 2W (Production system):

[octoprint-systeminfo-20230720173900-PROD.zip](https://github.com/OctoPrint/OctoPrint/files/12114597/octoprint-systeminfo-20230720173900-PROD.zip)

System bundle from Raspberry Pi 3B+ (Test system):

[octoprint-systeminfo-20230720173923-TEST.zip](https://github.com/OctoPrint/OctoPrint/files/12114604/octoprint-systeminfo-20230720173923-TEST.zip)

Example gcode file used for testing:

[Fish_Stick-Repair_Tube_0.2mm_PETG_MK3S_4m.gcode.gz](https://github.com/OctoPrint/OctoPrint/files/12114615/Fish_Stick-Repair_Tube_0.2mm_PETG_MK3S_4m.gcode.gz)

GordonW-Printer, 2023-07-20:
Sorry I renamed files to make them easier to identify:

Production:
[octoprint-systeminfo-20230720173900.zip](https://github.com/OctoPrint/OctoPrint/files/12114632/octoprint-systeminfo-20230720173900.zip)

Test:
[octoprint-systeminfo-20230720173923.zip](https://github.com/OctoPrint/OctoPrint/files/12114635/octoprint-systeminfo-20230720173923.zip)

cp2004, 2023-07-20:
This is intentional - the virtual printer is designed to simulate a communication error on line 100, 105, and 110 every time you use it. To test that the error checking & recovery works correctly. It's intended as a development tool in the first instance, so it has some pre-programmed stuff like this for testing.

closed 2023-07-20 by cp2004

GordonW-Printer, 2023-07-20:
Thanks for the response, is this behavior documented anywhere?

cp2004, 2023-07-21:
I'm not aware of it.

It's visible in the source code:

https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/plugins/virtual_printer/virtual.py#L414-L445

I guess nobody looked close enough at the output to consider it an 'issue' before.