OctoPrint/OctoPrint #4862: [BUG] After Filament Runout Sensor was triggered, gives `State: Canceling` when it is still actually printing normally.

classicrocker883, 2023-07-30:
### The problem

This happened direct after `Filament Runout Sensor` was triggered. It went into an `ADVANCED_PAUSE_FEATURE` for swapping filament.

Steps:
1. Start OctoPrint and Printer
2. Start a Print - observe normal behavior
3. Trigger Filament Runout Sensor  
<sub>(possible to hit "Pause" and BUG may happen then too - UNTESTED)</sub>
4. Observe OctoPrint UI State. Says "State: Canceling"

Should not say "Canceling" unless being canceled. The print, however, is still running normally.

May this be an issue with Mriscoc's ProUI? or OctoPrint?

[here is a link to the repo](https://github.com/classicrocker883/MriscocProUI) to My version of the ProUI Firmware  
[here is a linkt to Mriscoc's original firmware](https://github.com/mriscoc/Ender3V2S1)

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

currently involved in a print

### Version of OctoPrint

1.9.2

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Voxelab Aquila, MriscocProUI

### Browser and version of browser, operating system running browser

Chrome 115.0.5790.110

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Systeminfo Bundle [octoprint-systeminfo-20230729220900.zip](https://github.com/OctoPrint/OctoPrint/files/12208395/octoprint-systeminfo-20230729220900.zip)  
JavaScript browser console log [aquila.local-1690683134866.log](https://github.com/OctoPrint/OctoPrint/files/12208402/aquila.local-1690683134866.log)  
<br>

[image]

You can see in the screenshot it says "Canceling" when that is obviously not the case since it is still printing.

foosel, 2023-07-31:
> ### If you could not test in safe mode, please state why
> 
> currently involved in a print

As it says right in the issue form, "currently printing a print" is not an excuse to not test in safe mode! Please finish your print, then test in safe mode, then get back to us on whether the issue persists.

This is not to annoy you, this is because a lot of issues are caused by third party plugins which get disabled in safe mode, and debugging stuff that is not even part of core OctoPrint costs valuable time that is then not available for hunting actual core bugs or improving things in general for everyone.

foosel, 2023-08-28:
@classicrocker883 any chance on getting the requested test in safe mode from you? If not I'll have to close this.

classicrocker883, 2023-09-14:
hello I am currently having an issue with the printer. but I will be able to get everything we need very soon. thank you for being patient

foosel, 2023-10-02:
Can you define when "very soon" will be? 😅

classicrocker883, 2023-10-05:
finally just got everything up and running, technical issues... I almost forgot what the initial issue was or how it started 😳.
anyway I'll start a print and try triggering the thing again but in safe mode. is there anything else I would need or how to go about? such as recording any diagnosis, logging....?

foosel, 2023-10-05:
Enter safe mode, enable serial logging, (re)connect to your printer, reproduce the problem (if possible), immediately generate a systeminfo bundle and upload it here.

Safe mode: https://docs.octoprint.org/en/master/features/safemode.html

Serial logging: You can enable and disable serial.log via Settings > Serial Connection > General > Serial Logging.

Systeminfo bundle: https://community.octoprint.org/t/what-is-a-systeminfo-bundle-and-how-can-i-obtain-one/29887

classicrocker883, 2023-10-05:
unfortunately I wasn't able to recreate the issue. I tried with and without safemode. I do have the logging files but without triggering an issue that doesn't do us help? I am using a different gcode and the firmware has been updated since. I think I used a different slicer when the original problem occurred.

Ill have to go back to that file, slim it down so it hopefully triggers the code sooner rather than hours into a print.

just an M600 should cause the issue as I stated I initially.

has something like this happened to anyone at any other time? could it be then a one off thing since it doesn't seem to happen again?

jneilliii, 2023-10-05:
There was a marlin bug at one point in time if my vague memory serves me well, that when a filament change was triggered it wouldn't send the required host command back to OctoPrint when the process was resumed, etc. Of course, `HOST_ACTION_COMMANDS` also have to be enabled in the firmware for the process to have any chance of working at all.

classicrocker883, 2023-10-06:
> There was a marlin bug at one point in time if my vague memory serves me well, that when a filament change was triggered it wouldn't send the required host command back to OctoPrint when the process [...]

this is helpful info because in my testing, I havent been able to recreate the issue. I tried several ways.

it is possible this issue is fixed through an earlier commit.

otherwise is there anyway we'd be able to look at the previous logging files and find what we might be looking for, despite not being from SafeMode,or ect...?

xboxhacker, 2024-07-18:
I know this issue is a little old at this time, but I am having the same issue.

**The problem**
This happened directly after Filament Runout Sensor was triggered. It went into an ADVANCED_PAUSE_FEATURE for swapping filament.

**Steps:**
- Start OctoPrint and Printer
- Start a Print - observe normal behavior
- Trigger Filament Runout Sensor
- (possible to hit "Pause" and BUG may happen then too - UNTESTED)
- Observe OctoPrint UI State. Says "State: Cancelling" (see images)
- Should not say "Cancelling" unless being cancelled. The print, however, is still running normally.
- Cannot trigger a new:
- -Filament Runout
- -Pause
- -Cancel

**Did the issue persist even in safe mode?**
Tested in safe mode, same results

**If you could not test in safe mode, please state why**
N/A

**Version of OctoPrint**
Version 1.10.2

**Operating system running OctoPrint**
Octo4A 
Running on an Android phone.

**Printer model & used firmware incl. version**
TronXY VEHO600 (basically an ender)
MarlinFW 2.1.1
OctoEverywhere 

**Marlin Snippets**
`#define HOST_ACTION_COMMANDS
#if ENABLED(HOST_ACTION_COMMANDS)
  #define HOST_PAUSE_M76                // Tell the host to pause in response to M76
  #define HOST_PROMPT_SUPPORT           // Initiate host prompts to get user feedback
  #if ENABLED(HOST_PROMPT_SUPPORT)
    #define HOST_STATUS_NOTIFICATIONS   // Send some status messages to the host as notifications
  #endif
  #define HOST_START_MENU_ITEM          // Add a menu item that tells the host to start
  #define HOST_SHUTDOWN_MENU_ITEM       // Add a menu item that tells the host to shut down
#endif`

`#define ADVANCED_PAUSE_FEATURE
#if ENABLED(ADVANCED_PAUSE_FEATURE)
  #if TAOLI_SERIAL
  #define PAUSE_PARK_RETRACT_FEEDRATE         2
  #define PAUSE_PARK_RETRACT_LENGTH           2
  #else
  #define PAUSE_PARK_RETRACT_FEEDRATE         20  // (mm/s) Initial retract feedrate.
  #define PAUSE_PARK_RETRACT_LENGTH            2  // (mm) Initial retract.
                                                  // This short retract is done immediately, before parking the nozzle.
  #endif
  #define FILAMENT_CHANGE_UNLOAD_FEEDRATE     10  // (mm/s) Unload filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_UNLOAD_ACCEL        25  // (mm/s^2) Lower acceleration may allow a faster feedrate.
  #define FILAMENT_CHANGE_UNLOAD_LENGTH      100  // (mm) The length of filament for a complete unload.
                                                  //   For Bowden, the full length of the tube and nozzle.
                                                  //   For direct drive, the full length of the nozzle.
                                                  //   Set to 0 for manual unloading.
  #if TAOLI_SERIAL
  #define FILAMENT_CHANGE_SLOW_LOAD_FEEDRATE   2
  #else
  #define FILAMENT_CHANGE_SLOW_LOAD_FEEDRATE   6  // (mm/s) Slow move when starting load.
  #endif
  #define FILAMENT_CHANGE_SLOW_LOAD_LENGTH     0  // (mm) Slow length, to allow time to insert material.
                                                  // 0 to disable start loading and skip to fast load only
[... 30 lines cut ...]
  #define PAUSE_PARK_NO_STEPPER_TIMEOUT           // Enable for XYZ steppers to stay powered on during filament change.
  //#define FILAMENT_CHANGE_RESUME_ON_INSERT      // Automatically continue / load filament when runout sensor is triggered again.
  //#define PAUSE_REHEAT_FAST_RESUME              // Reduce number of waits by not prompting again post-timeout before continuing.

  #define PARK_HEAD_ON_PAUSE                    // Park the nozzle during pause and filament change.
  //#define HOME_BEFORE_FILAMENT_CHANGE           // If needed, home before parking for filament change

  //#define FILAMENT_LOAD_UNLOAD_GCODES           // Add M701/M702 Load/Unload G-codes, plus Load/Unload in the LCD Prepare menu.
  //#define FILAMENT_UNLOAD_ALL_EXTRUDERS         // Allow M702 to unload all extruders above a minimum target temp (as set by M302)
#endif
`

**Browser and version of browser, operating system running browser**
Chrome Version 126.0.6478.127 (Official Build) (64-bit)

**Files**
[octoprint.log](https://github.com/user-attachments/files/16285497/octoprint.log)

[image]
[image]
[image]

I would like to point out that I am new to Octoprint, but not new to printing.

Thanks!

xboxhacker, 2024-07-18:
In the middle of a 24-hour print, I will update later with my bundle download....

jneilliii, 2024-07-18:
So based on one of the screenshots you are printing from SD card instead of from OctoPrint. For best assistance, serial logging needs to be enabled while the issue is happening before creating a system info bundle.

xboxhacker, 2024-07-18:
I will enable it, and report back.

Side note: Printing from the SD card is the **_ONLY_** way for me to get the machine to pause on a Filament Runout. Printing from the UI does not trigger a pause.

xboxhacker, 2024-07-19:
Here is my bundle file, with serial.log enabled.

[octoprint-systeminfo-20240719141858.zip](https://github.com/user-attachments/files/16313323/octoprint-systeminfo-20240719141858.zip)

`2024-07-19 14:14:15,490 - Recv: //action:prompt_begin FilamentRunout T0
2024-07-19 14:14:15,495 - Recv: //action:prompt_show
2024-07-19 14:14:15,505 - Recv: //action:filament_runout T0
2024-07-19 14:14:15,510 - Recv: //action:pause filament_runout 0
2024-07-19 14:14:15,513 - Pausing on request of the printer...
2024-07-19 14:14:15,513 - Changing monitoring state from "Printing from SD" to "Pausing"
2024-07-19 14:14:15,529 - Send: M25
2024-07-19 14:14:15,534 - Recv: X:315.78 Y:75.81 Z:0.60 E:55.54 Count X:16840 Y:4043 Z:160
2024-07-19 14:14:15,561 - Recv: SD printing byte 1440/296698
2024-07-19 14:14:16,544 - Recv:  T:260.44 /260.00 B:80.49 /80.00 @:56 B@:0
2024-07-19 14:14:16,563 - Recv: SD printing byte 1440/296698
2024-07-19 14:14:17,206 - Recv: echo:busy: processing
2024-07-19 14:14:17,209 - Recv: X:314.40 Y:70.69 Z:0.60 E:55.54 Count X:16767 Y:3770 Z:160
2024-07-19 14:14:17,215 - Recv: S_XYZ:5
2024-07-19 14:14:17,560 - Recv: SD printing byte 1440/296698
2024-07-19 14:14:18,546 - Recv:  T:261.22 /260.00 B:80.39 /80.00 @:40 B@:0
2024-07-19 14:14:18,563 - Recv: SD printing byte 1440/296698
2024-07-19 14:14:19,204 - Recv: echo:busy: processing
2024-07-19 14:14:19,207 - Recv: X:305.10 Y:65.12 Z:0.60 E:55.54 Count X:16271 Y:3473 Z:160
2024-07-19 14:14:19,216 - Recv: S_XYZ:5
2024-07-19 14:14:19,560 - Recv: SD printing byte 1440/296698
2024-07-19 14:14:19,847 - Recv: S_XYZ:5
2024-07-19 14:14:19,849 - Recv: S_XYZ:5
2024-07-19 14:14:19,850 - Recv: S_XYZ:5
2024-07-19 14:14:19,854 - Recv: S_XYZ:5
2024-07-19 14:14:19,857 - Recv: S_XYZ:5
2024-07-19 14:14:19,859 - Recv: S_XYZ:5
2024-07-19 14:14:19,862 - Recv: S_XYZ:5
2024-07-19 14:14:20,054 - Recv: echo:Unknown command: "M5"
2024-07-19 14:14:20,058 - Recv: //action:paused
2024-07-19 14:14:20,061 - Printer signalled that it paused, switching state...
2024-07-19 14:14:20,766 - Recv: X:315.77 Y:75.81 Z:20.00 E:53.88 Count X:16840 Y:4043 Z:5333
2024-07-19 14:14:20,796 - Recv:  T:261.46 /260.00 B:80.31 /80.00 @:39 B@:0
2024-07-19 14:14:20,805 - Recv: Not SD printing
2024-07-19 14:14:21,789 - Recv: Not SD printing
2024-07-19 14:14:22,045 - Recv: echo:busy: processing
2024-07-19 14:14:22,048 - Recv: X:315.77 Y:75.81 Z:16.81 E:53.88 Count X:16840 Y:4043 Z:4483
2024-07-19 14:14:22,061 - Recv: S_XYZ:5
2024-07-19 14:14:22,776 - Recv:  T:261.41 /260.00 B:80.19 /80.00 @:45 B@:0
2024-07-19 14:14:22,789 - Recv: Not SD printing
2024-07-19 14:14:22,791 - Changing monitoring state from "Pausing" to "Cancelling"
2024-07-19 14:14:22,822 - Send: M108
2024-07-19 14:14:23,791 - Recv: Not SD printing`

It continues if I select resume, but it is now locked out of all control including any new filament runouts.

xboxhacker, 2024-07-19:
But even with the "M5" error, it received.

`Recv: //action:paused`

Need to find out where the M5 is being sent from? Disable/surpress it somehow?

classicrocker883, 2024-07-24:
> But even with the "M5" error, it received.
> 
> `Recv: //action:paused`
> 
> Need to find out where the M5 is being sent from? Disable/surpress it somehow? There is no reference to M5 in GCODE file.

M5: Turn OFF Laser | Spindle

Is `PARK_HEAD_ON_PAUSE` enabled?

xboxhacker, 2024-07-24:
Is `PARK_HEAD_ON_PAUSE` enabled? **Yes it is.**
I have tested `M125` and it works as it should. Actually, had put this into octoprint GCODE, it was the only way to get it to park.
Got a few more things to try, but I am running out of ideas.

Here is my main issue. The place with the printer has no internet. So I am using a Moto One 5g with Octo4A and Octoeverywhere. This way I can monitor (also with AI) my prints, which are all over 24hr ea. So I cannot use a Pi like a normal setup. I need to print from the SD because if I print from the UI octoprint "stutters" and the print gets jacked up.

Even if OEW triggers a AI problem and needs to pause, it goes into canceling.

xboxhacker, 2024-07-25:
Latest bundle file....

[octoprint-systeminfo-20240725143520.zip](https://github.com/user-attachments/files/16379825/octoprint-systeminfo-20240725143520.zip)

foosel, 2024-08-12:
The problem is that the firmware is implementing the `SD printing` messaging wrong here.

```
2024-07-19 14:14:20,058 - Recv: //action:paused
2024-07-19 14:14:20,061 - Printer signalled that it paused, switching state...
2024-07-19 14:14:20,766 - Recv: X:315.77 Y:75.81 Z:20.00 E:53.88 Count X:16840 Y:4043 Z:5333
2024-07-19 14:14:20,796 - Recv: T:261.46 /260.00 B:80.31 /80.00 @:39 B@:0
2024-07-19 14:14:20,805 - Recv: Not SD printing
2024-07-19 14:14:21,789 - Recv: Not SD printing
2024-07-19 14:14:22,045 - Recv: echo:busy: processing
2024-07-19 14:14:22,048 - Recv: X:315.77 Y:75.81 Z:16.81 E:53.88 Count X:16840 Y:4043 Z:4483
2024-07-19 14:14:22,061 - Recv: S_XYZ:5
2024-07-19 14:14:22,776 - Recv: T:261.41 /260.00 B:80.19 /80.00 @:45 B@:0
2024-07-19 14:14:22,789 - Recv: Not SD printing
2024-07-19 14:14:22,791 - Changing monitoring state from "Pausing" to "Cancelling"
```

On filament runout, the printer pauses the print, so far so good. However, the firmware now starts reporting that it is `Not SD printing`. That's wrong. It should be reporting that it in fact still IS SD printing - it's still in the middle of an ongoing SD print job, it only is waiting for a filament swap to happen before it can continue. The firmware should be reporting that it is still `SD Printing`, with a static byte position in the still open file. What it does instead is signaling that it no longer is printing at all, and OctoPrint thus tries to mirror that state and switches over to `Cancelling`.

So - this is a firmware bug. The question is how widespread this is and if it made it into some mainline firmware code, or if this is only happening on vendor forks. Was this version of Marlin compiled directly from Marlin's Github or where did it come from?

Side note: the firmware is also doing seriously some serious and completely unnecessary prompt spamming here. The prompt feature was provided to allow firmware to *interact* with the user, when she needs to make a decision. It was never supposed to be abused for constantly spamming the user with state switches that need to be acknowledged in the prompt. It's frustrating to see this feature abused like that.

foosel, 2024-08-13:
> Was this version of Marlin compiled directly from Marlin's Github or where did it come from?

@xboxhacker That's a question for you.

I've just spent a couple of hours going through the official Marlin 2.1.x code and so far I cannot confirm this behaviour in it - from as far as I understand the flow, a filament run out will trigger some commands to be enqueued: `M25` to pause the print, which in turn will trigger an `M125` if support for that is configured to park the nozzle. I have not seen a reference to `M5` and that also wouldn't make a whole ton of sense in this context (and require `HAS_CUTTER` to work).

If this *is* an official Marlin release from [`MarlinFirmware/Marlin`](https://github.com/MarlinFirmware/Marlin), I think at this point I'd like to get my hands on the configuration files you used, maybe that will allow me to figure out what the heck is going on there with regards to the broken SD printing status which causes this mess.

If this is not an official release - use one instead of whatever this is? 😅

---

As a side note, there's a ton of "unknown command" errors regarding `M73`, which is used to set the printer progress. The firmware reports `Cap:PROGRESS:0`, so this is not supported. Either a plugin is injecting that stuff, or it's inside your compiled GCODE, but in any case, you need to enable progress support or change/disable whatever puts them in, they don't make sense here. I don't think that's related however.

xboxhacker, 2024-11-01:
So for the delay! This is the official firmware from TronXY GH. The machine is TronXY VEHO 600.

The zip file is my **full PIO** marlin folder.
[VEHO FIRMWARE.zip](https://github.com/user-attachments/files/17600114/VEHO.FIRMWARE.zip)

foosel, 2024-11-01:
So it's a vendor fork. In that case please report this firmware bug to TronXY - they broke something in there, I've outline what above, and if they need help in fixing it they are welcome to contact me about consulting. But I can't implement worksheets for every broken vendor fork under the sun sadly - there are far more vendors breaking things than there are copies of me 😂

closed 2024-11-01 by foosel