OctoPrint/OctoPrint #4863: Webcam Stream Loading... Safari

KaJashey, 2023-07-30:
### The problem

The problem: Octoprint 1.9.2 and earlier. Intermittently when viewing the control tab in Safari I get a message "Webcam Stream Loading..." and the video does not load.

To fix the problem: Clear cache and restart Safari:

To cause the problem to reoccur: Open two browser tabes with the Octoprint: Control tab open. The second browser tab will have the "Webcam Stream Loading..." message while the first plays video. Closing those two browser tabs and opening another gets the "Webcam Stream Loading..." message. This persists till cache is cleared and browser is restarted.

This doesn't not happen on Chrome: Chrome can have two browser tabs with the control window open and video streams to each.

Will try to report to Apple/Safari team as well

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why

Currently printing a big print

### Version of OctoPrint

1.9.2

### Operating system running OctoPrint

Linux

### Printer model & used firmware incl. version

 Prusa Mk3S+ - Firmware 3.13.0 

### Browser and version of browser, operating system running browser

Safari 16.5.2 /605.1.15 Mac OS Ventura 13.4.1(c)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Systeminfo Bundle:
browser.user_agent: Mozilla/5.0 (Macintosh; Intel Mac OS X 10_15_7) AppleWebKit/605.1.15 (KHTML, like Gecko) Version/16.5.2 Safari/605.1.15
connectivity.connection_check: 8.8.8.8:53
connectivity.connection_ok: True
connectivity.enabled: True
connectivity.online: True
connectivity.resolution_check: octoprint.org
connectivity.resolution_ok: True
env.hardware.cores: 4
env.hardware.freq: 1512.0
env.hardware.ram: 1031921664
env.os.bits: 64
env.os.id: linux
env.os.platform: linux
env.python.pip: 22.3.1
env.python.version: 3.9.2
env.python.virtualenv: True
octoprint.last_safe_mode.date: unknown
octoprint.last_safe_mode.reason: unknown
octoprint.safe_mode: False
octoprint.version: 1.9.2
printer.firmware: Prusa-Firmware 3.13.0 based on Marlin
systeminfo.generated: 2023-07-30T14:01:31Z
systeminfo.generator: zipapi

foosel, 2023-07-31:
> ### If you could not test in safe mode, please state why
> 
> Currently printing a big print

As it says *right in the issue form*, "currently printing a big print" is **not** an excuse to not test in safe mode! Please finish your big print, then test in safe mode, then get back to us on whether the issue persists. 

This is not to annoy you, this is because a lot of issues are caused by third party plugins which get disabled in safe mode, and debugging stuff that is not even part of core OctoPrint costs valuable time that is then not available for hunting actual core bugs or improving things in general for everyone.

wolfgangrumpf, 2023-09-25:
I actually am having a problem that may be related - webcam was working fine then just stopped working; if I run **sudo service camera-streamer restart** it typically fixes it, at least for a while....

foosel, 2023-10-02:
@wolfgangrumpf that rather sounds like ayufan/camera-streamer#94

foosel, 2024-08-19:
Closing as there's been no additional information provided in a year by the OP, and it's likely a Safari issue in either case and thus out of scope here.

closed 2024-08-19 by foosel