OctoPrint/OctoPrint #4865: [Request] Support displaying Marlin SOC temp on graph

ansonl, 2023-08-01:
### Is your feature request related to a problem? Please describe.

Right now Marlin SOC temp can only be read in the terminal as the temperature reported as S:XX.XX. It's hard to track it during printing and over time.

### Describe the solution you'd like

Display SOC temp as another temperature in Octoprint like Chamber temp,etc.

### Describe alternatives you've considered

Log M105 temps that have S before them, but it's basically redoing the graph. 

### Additional context

https://github.com/MarlinFirmware/Marlin/pull/25642 implemented the ability to read SOC temperature.

jneilliii, 2023-08-01:
This can be accomplished with my Plotly Temp Graph plugin. Pretty sure it will report anything that comes across the response from M105.

kantlivelong, 2023-08-01:
Top Temp could also be a viable option if you're open to it being displayed in it's own view.

ansonl, 2023-08-01:
Top Temp worked for me since I'm already using the Tempsgraph plugin for a zoomable and croppable temp graph on the main temp tab. 

For anyone else finding this, SoC temp can be set up in Top Temp with Type `From printer` and Regexp `S:(\d*.\d*)`

closed 2023-08-01 by ansonl

jneilliii, 2023-08-02:
For reference, Plotly Temp Graph plugin is not the same as tempsgraph, but did have all those features as well.