OctoPrint/OctoPrint #4867: Old webcam settings system still used

cp2004, 2023-08-03:
### The problem

Following https://community.octoprint.org/t/warning-message/53242/9 I searched the source to find this one line that is still looking for the 'old' way of getting a snapshot URL:

https://github.com/OctoPrint/OctoPrint/blob/ed4a2646fb4e2904892c895580192987242834c8/src/octoprint/server/__init__.py#L934

Could be the one triggering the scary warnings confusing people about what they need to do...?

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.2

### Operating system running OctoPrint

N/A

### Printer model & used firmware incl. version

N/A

### Browser and version of browser, operating system running browser

N/A

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Systeminfo bundle available in the linked forum post - I just located this through searching not actually running OctoPrint.

foosel, 2023-08-10:
Fixed by the above commit. Will definitely be part of 1.10.0, but I might also push it out with a 1.9.x bugfix release, should another one become necessary.

cociweb, 2023-09-13:
Is it addresses that issue, where the Classic Webcam plugin cannot validate the snapshot url meanwhile both the stream and the snapshot itself is OK with the octoprint-docker default (haproxy) config? -> it results also in that timelaps feature not working.
If not, then should I open a new discussion/bug report?

commit 9b2faaaa (2023-08-10): 🐛 Switch to dedicated webcam snapshot handler

Fixes deprecation warning logged when creating tornado endpoint
/downloads/camera/current.

Closes #4867
closed 2024-04-24 by foosel