OctoPrint/OctoPrint #4869: [Request] Support events with multiple triggers in the web UI

mbudnek, 2023-08-06:
### Describe the solution you'd like

It's currently possible to define events with multiple triggers in config.yml, i.e.

```
events:
  subscriptions:
  - command: do_something
    debug: false
    enabled: true
    event:
    - PrintDone
    - PrintFailed
    - PrintCancelled
    name: My Event Subscription
    type: system
```

This same functionality is not available in the web UI's Settings->Octoprint->Event Magager page.  In fact, such events cause that page to fail to populate entirely with the error:

```
2023-08-06 03:08:24,277 - octoprint.server.api.settings - ERROR - Could not load settings for plugin Event Manager (None)
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/settings.py", line 431, in _get_plugin_settings
    result = plugin.on_settings_load()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugins/eventmanager/__init__.py", line 24, in on_settings_load
    events.get("subscriptions", []), key=(lambda x: x["event"])
TypeError: '<' not supported between instances of 'str' and 'list'
```

It would be very nice to be able to define and edit these multi-trigger events from the web UI instead of having to modify config.yml manually.

jneilliii, 2023-08-06:
This has already been identified and https://github.com/OctoPrint/OctoPrint/pull/4851 created to address the issue.

foosel, 2023-08-10:
Implemented by the above commit and will be rolled out with 1.10.0

foosel, 2024-04-25:
1.10.0 has been released.

closed 2024-04-25 by foosel