OctoPrint/OctoPrint #4871: Temperature graph tab missing

Siafu7677, 2023-08-11:
### The problem

1. Temperature graph has disappeared but control tab still functions

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.2

### Operating system running OctoPrint

Windows 11

### Browser and version of browser, operating system running browser

Firefox v116.0.2 64b

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230811123047.zip](https://github.com/OctoPrint/OctoPrint/files/12323463/octoprint-systeminfo-20230811123047.zip)
[Firefox Console.txt](https://github.com/OctoPrint/OctoPrint/files/12323521/Firefox.Console.txt)
[image]

foosel, 2023-08-11:
According to your bundle you last started in safe mode in February 🤨 

So, please test again in safe mode and provide a screenshot from THAT if the problem persists.

Also reenable the throttle check in the Pi Support plugin which you currently have disabled according to your bundle. It might just as well be an issue caused by undervoltage or overheating.

Siafu7677, 2023-08-11:
[Firefox Console.txt](https://github.com/OctoPrint/OctoPrint/files/12324316/Firefox.Console.txt)
[octoprint-systeminfo-20230811145505.zip](https://github.com/OctoPrint/OctoPrint/files/12324317/octoprint-systeminfo-20230811145505.zip)

Siafu7677, 2023-08-11:
Safe mode it shows up, new console and sysinfo included for review

closed 2023-08-11 by Siafu7677

Siafu7677, 2023-08-11:
Incompatible plug-in was the cause since turning off TempsGraph made it show up. WIll post issue to that plug-ins GIT

jneilliii, 2023-08-11:
If you want a similar graph as that one that is functional try out my Plotly Temp Graph plugin.

foosel, 2023-08-12:
Great this got resolved. So next time, instead of just claiming to have tested whether the issue persists in safe mode, *actually test whether the issue persists in safe mode*. Safe mode is precisely for situations like this one.