OctoPrint/OctoPrint #4872: la connexion USB du laser ne fonctionne plus avec la version 1.92 de octoprint

DU4FDR, 2023-08-11:
### The problem

Suite à la mise à jour de la version 1.92 la connexion des laser ne fonctionne plus !
Le port USB donne des erreurs 09 ou 0.20.
à priori je pense rétrograder vers la version 1.87.
Comment faire cette rétroaction, merci d'avance de votre aide !
à priori c'est un problème connu mais pas de correctif pour le plugin GRBL...

[image]
[image]

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.92

### Operating system running OctoPrint

OctoPrint 1.9.2  Python 3.9.2  OctoPi* 1.0.0 (build 2023.07.18.111854)

### Printer model & used firmware incl. version

laser TWO TREES

### Browser and version of browser, operating system running browser

edge

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

J'ai réalisé la connexion sur le port USB avec une imprimante çà fonctionne normalement !

Si je connecte le laser le système me renvoie une erreur sur le firmware du laser ?
je ne sais plus quoi faire ?

cp2004, 2023-08-26:
This is a known issue with the Better GRBL plugin - it's not compatible with OctoPrint 1.9.0 yet. See here https://github.com/synman/Octoprint-Bettergrblsupport/issues/118

closed 2023-08-26 by cp2004

synman, 2023-10-11:
I know this is closed but the issue in the plugin does suggest regression within 1.9 tied to profile management.  There were some big changes in 1.9 that broke backwards compatibility related to profiles.

DU4FDR, 2023-10-12:
Good morning,
Do you plan to correct the problem of using lasers with OCTOPRINT which do not currently work with version 1.9xx? we are forced to use version 1.87 with a minimum of options for GRBL commands.
please inform us because we use the enclosure module.
Octoprint allows you to safely control a laser and protect yourself from radiation and dust. !
a+ Francis

----- Mail original -----

foosel, 2023-10-12:
The problem lies with a 3rd party plugin that is not maintained by me. As far as I understand, the problem stems from profile validation introduced in 1.9.x that fixed existing bugs and has already been fixed in the source of the plugin, it only needs to be turned into a new release.

If there are further issues, I'm happy to help the maintainer in solving them, and if it turns out to be due to a regression, I'm also happy to fix stuff on my end.