OctoPrint/OctoPrint #4873: Octoprint hanging on "heating" of hotend

averter, 2023-08-17:
### The problem

1. I was trying to extrude/retract filament after heating PLA up to 203ºC
2. I've set the hotend temp to 203ºC in the octoprint temperature menu and it started heating normally
3. After reaching the target temp it fluctuated a bit as usual, BUT strangely, the lcd control kept displaying "Heating..."
4. The printer remained in this state and I could not extrude/retract filament. I can make it move, turn the hotend off, etc. but that's about it.
5. I've checked and the bed is fine (only the hotend seems afflicted by the bug). I've also tried to set the temperature offset to 2ºC but it did not help. 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.2

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

Prusa i3 rework Marlin 1.1.0-RC6 

### Browser and version of browser, operating system running browser

Firefox, Linux mint

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230817081114.zip](https://github.com/OctoPrint/OctoPrint/files/12366492/octoprint-systeminfo-20230817081114.zip)

sarusani, 2023-08-23:
Sounds like a problem with your printer/firmware. Have you tried flashing a different version and see if the error persists?

Octoprint just tells the printer to heat to a certain degree. It's the printer who decides if the requested temp has been reached.

averter, 2023-08-23:
The reason why I think it's octoprint related it's because I've recently updated to 1.9.2 and it has been since then that this issue happened. I can try to bisect the problem by reverting back to previous versions of octoprint as well.

cp2004, 2023-08-26:
I think I'd agree with @sarusani's assessment - if you ask the printer to heat up with a blocking heating command, it doesn't let OctoPrint send any more commands until it's happy the temperature has stabilised.

You most likely just need to do a PID tune to get the temps stable - if they are fluctuating, then the printer won't decide it's stable enough to hand back control.

averter, 2023-08-26:
Hi @cp2004 The temperature is stable as far as I can tell from the graph/numbers in the lcd display. Another strange thing is that this issue only happens when I heat the hotend manually, i.e., if I print something the printer will heat the bed + hotend and start extruding normally. 
I'll try to revert to another version of Marlin and see if the issue disappears and report back.

cp2004, 2023-08-26:
In that case, if you could enable serial.log, reproduce the issue and share either just the log or a systeminfo bundle again so we can see the communication to the printer.

averter, 2023-09-11:
Here it is
[serial.log](https://github.com/OctoPrint/OctoPrint/files/12578569/serial.log)

jneilliii, 2023-09-11:
I wonder if this similar to the other issue related to extrusion, #4849.

averter, 2023-09-18:
> I wonder if this similar to the other issue related to extrusion, #4849.

Bingo. After adding `("axes", "e", "speed"),` to `/home/pi/oprint/lib/python3.9/site-packages/octoprint/printer/profile.py` it is now working. So this is a duplicate. Thanks!

closed 2023-09-18 by averter