OctoPrint/OctoPrint #4875: After changing extruder speed, extrude button no longer extrudes. After changing speed back, original speed no longer works.

johnscott4, 2023-08-25:
### The problem

I find that on a fresh install of octoprint, the extrude button works as expected. After changing the extruder speed to 100, from the default of 300, the extrude button no longer appears to send any command. After changing back to 300, it still does not work. Re-installing octoprint, it works again at the default 300, but after changing it again, it breaks once more.

Looking at the logs, I see a python error:
    ["G91", "M83", "G1 E%s F%d" % (amount, extrusion_speed), "M82", "G90"],
TypeError: %d format: a number is required, not str

Appears to be a bug, but seems strange i'm the only one seeing this.

Please let me know if you need any more information. 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.2

### Operating system running OctoPrint

OctoPi* 1.0.0 (build 2023.07.18.111854)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint (1).log](https://github.com/OctoPrint/OctoPrint/files/12435912/octoprint.1.log)
[octoprint-systeminfo-20230825063612.zip](https://github.com/OctoPrint/OctoPrint/files/12435932/octoprint-systeminfo-20230825063612.zip)

sarusani, 2023-08-26:
Duplicate of #4849

jneilliii, 2023-08-26:
It's this duplicate of #4849 ?

foosel, 2023-08-28:
It is.

Closing as duplicate of #4849