OctoPrint/OctoPrint #4876: Upgrade failed

dheera, 2023-08-27:
### The problem

Upgrade failed

```

2023-08-27 03:37:41,051 - octoprint.plugins.softwareupdate - ERROR - Update of pi_support can not be performed, please also check plugin_softwareupdate_console.log for possible causes of this
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 1828, in _perform_update
    target, populated_check, target_version, log_cb=self._log, online=online
  File "/home/pi/oprint/lib/python2.7/site-packages/octoprint/plugins/softwareupdate/updaters/pip.py", line 144, in perform_update
    "Error while executing pip install", (stdout, stderr)
UpdateError: Error while executing pip install
2023-08-27 03:37:41,713 - octoprint.plugins.tracking - INFO - Sent tracking event update_failed, payload: {'to_version': u'2023.7.31', 'from_version': '2021.6.14', 'target': u'pi_support'}
```
Logs attached

[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/12447184/octoprint.log)
[plugin_softwareupdate_console.log](https://github.com/OctoPrint/OctoPrint/files/12447185/plugin_softwareupdate_console.log)
[plugin_pluginmanager_console.log](https://github.com/OctoPrint/OctoPrint/files/12447186/plugin_pluginmanager_console.log)

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.6.1 -> 1.9.2

### Operating system running OctoPrint

OctoPi

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

foosel, 2023-08-28:
> octoprint.version: 1.6.1
> env.python.version: 2.7.16

Python 2 has been end of life since the start of 2020, OctoPrint has not been supporting it since version 1.8.0 released in May of 2022. Neither do its bundled plugins now. 

You are on an ancient unsupported version. [Finally move to Python 3 please](https://octoprint.org/blog/2022/01/31/octoprint-1.8.0-will-require-python-3/), as repeatedly mentioned in FAQ, Blog, OctoPrint on Air and every other place over the past years.

closed 2023-08-28 by foosel

dheera, 2023-08-31:
Thanks! I didn't realize this. (I thought the "upgrade" button would deal with dependencies such as Python versions. Might be a suggestion for future to containerize it with conda or docker and allow OctoPrint to upgrade its own Python.)