OctoPrint/OctoPrint #4877: Webcam preview breaks if any tab other than "command and control" is default

WimVDK, 2023-08-27:
### The problem

Running octoprint 1.9.2 and all plugins up-to-date. Having any tab other than C&C enabled breaks the preview (which is configured to connect to a Wyzecam v2 running openmiko), webcam stream url itself still works fine and preview works perfectly fine in the "dashboard" plugin i installed to test.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

webcam gets disabled in safe mode

### Version of OctoPrint

1,9,2

### Operating system running OctoPrint

OctoPi 0.18.0

### Printer model & used firmware incl. version

CR10 Max, running bondtech firmware CR10Max_50W_ET_DW7.4.7

### Browser and version of browser, operating system running browser

Firefox 116.0.3

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230827203036.zip](https://github.com/OctoPrint/OctoPrint/files/12448592/octoprint-systeminfo-20230827203036.zip)

jneilliii, 2023-08-27:
> webcam gets disabled in safe mode

Are you sure about this? The classic webcam plugin is bundled and shouldn't get disabled in safe mode.

WimVDK, 2023-08-27:
> Are you sure about this

If it should be present on the main UI, yes. Unless I'm doing something wrong (clicked on the na bar and Selected 'boot to safe mode').

cp2004, 2023-08-27:
Can you share screenshots of where you see the webcam & where you'd expect it to be - it should be on the control tab, definitely in safe mode.

WimVDK, 2023-08-28:
I'll do one better. Made a short video. (3:48m) Anything but being on the main tab breaks the webcam preview? Not 100% sure now if this is a ui customizer thing or octoprint since webcam did work on safe mode. The preview window was gone so i thought it wasnt.

[Google drive link](https://drive.google.com/file/d/1VJ14N_tg6PENQ3S4E6MOSIkrwy_nujO4/view?usp=sharing)

jneilliii, 2023-08-28:
You sure this isn't an issue with Consolidated Temp Control plugin? If you disable that plugin and go back to the regular Control tab on its own does the same thing happen?

closed 2023-08-28 by cp2004

WimVDK, 2023-08-28:
Reported to UICustomizer too then :) It's not a consolidated temp control thing. If it were the webcam "command and control" would break too :)