OctoPrint/OctoPrint #4878: [Request]

EmEiBee, 2023-08-28:
### Is your feature request related to a problem? Please describe.

The problem has already been described here (https://community.octoprint.org/t/storing-snaps-and-other-stuff-closed-solved/53560/12).
Short version:
1.
If you move the directories for the uploads, timelapse, etc. from the SD card via mount to protect them, the API to CURA no longer works.
2.
If OctoPrint is faster than the mount can already be set, OctoPrint switches back to the default paths without being asked.

### Describe the solution you'd like

1.
I have no idea how the API works (absolutely no idea about coding), but it would be nice if the API worked on a mount too.
2.
That, in my humble opinion, should at least be reported clearly. 
However, it would be better if you could define a timeout here by option/check mark, for example, how long to wait for the mount to be provided and if failed fall back to default paths and inform the user

foosel, 2023-08-29:
Please do not create one issue for multiple things.

> If you move the directories for the uploads, timelapse, etc. from the SD card via mount to protect them, the API to CURA no longer works.

"no longer works" can mean a lot of things, and it's not the API to Cura, it's OctoPrint's general REST API. It stands to reason whether this is a Cura or OctoPrint problem, but for that you've not provided enough information. Please open a proper bug report with all requested information.

> If OctoPrint is faster than the mount can already be set, OctoPrint switches back to the default paths without being asked.

In that case you should be adjusting the unit file starting OctoPrint to wait for your mount to be ready. That has already been explained to you on the forums as well. OctoPrint tries its best to be fault tolerant here so we don't drown in support requests from people messing around with stuff and then complaining that the server does no longer start up (that used to be the case before). 

> That, in my humble opinion, should at least be reported clearly.

It is stated quite aggressively in the log. If you mess around with mount points, I think it's fair to expect that you also would be looking at OctoPrint's logs to verify your changes work and are not causing issues. I can look into a warning on the UI however.

EmEiBee, 2023-08-29:
... just forget it then ...

foosel, 2023-08-29:
Ah, how lovely. Listen, I WANT to fix stuff, but I'm one person, and OctoPrint has hundreds of thousands of users. I can't run after every single issue trying to pry enough information from whoever is encountering it to fix it, the community (and that also means YOU) has to help at least with usable bug reports.

Linking to a confusing forum thread, jumbling various issues into one ticket without even bothering to add a title but still expecting me to be able to solve this is just expecting miracles.

This is not how open source works. Either you help me help you, or this won't get fixed. It's your choice, I won't lose sleep over it. But it's really disappointing to see your reaction here when asked to do the bare minimum of filling out a bug report to enable someone to solve your problem for you.

EmEiBee, 2023-08-31:
... what's wrong?
You say yourself it's not a BUG, then later you say it's one.
And at the latest after your statement "...It is stated quite aggressively in the log. If you mess around with mount points, I think it's fair to expect that you also would be looking at OctoPrint's logs to verify your changes work and are not causing issues...." I realized it's better to just forget about it.
Ultimately you assume that all users are the top programmers and know exactly what an API does, how it does it and why and also assumes that everyone here knows exactly why something works or not. I'm not a coder and just jumped into 3D-Stuff 4 weeks ago; can't be a professional coder like you in that short time ...
Of course I can partly understand you, because it happens to me in my area of ​​competence that I assume knowledge from third parties that they cannot have at all... But when I see, among other things, that a problem that has been known for years (I won't say BUG now), such as the discrepancy in the manual operation of the Y-axis in OctoPrint versus OctoDash is still there then I have no hope that something will be fixed that apparently hasn't been tried by anyone. So I'll do a backup of the SD card every few days and live with the problem...

cp2004, 2023-08-31:
The API issue uploading to Cura has always been referenced as a possible bug, but without a proper bug report (that includes a systeminfo bundle) we can't figure out where the problem lies. So, if you want it looked into/fixed => bug report.

Showing a warning in the UI has been turned into a new feature request, so everything looks alright there from what I can see?

> problem that has been known for years (I won't say BUG now), such as the discrepancy in the manual operation of the Y-axis in OctoPrint versus OctoDash is still

I am not aware of this problem. Could you link to the bug report so I can understand it?

foosel, 2023-08-31:
> You say yourself it's not a BUG, then later you say it's one.

That's because you put two separate things into ONE issue and then apparently forgot yourself about there being two separate things in one issue here and get confused when I make that distinction. And this is *precisely* why you should never put multiple things into one single issue.

The Cura problem you are observing (point 1 in your original post) could be a bug in OctoPrint or in Cura. I don't know. For me to be able to determine that, I need a proper [bug report](https://github.com/OctoPrint/OctoPrint/issues/new?assignees=&labels=&projects=&template=bug_report.yml).

A mount point having to be populated at server start time however is not a bug (point 2 in your original post), that's a configuration issue that can easily solved by doing what was explained to you in the forum thread - adjusting the unit file that starts the service to *wait for the mounts to be available*. I agreed that better messaging that OctoPrint falls back on known defaults in such a case is needed and created #4881 for that reason.

So I ask *you*, what's wrong with that that made you respond with a passive aggressive "... forget it then..." and now make utterly wrong assumptions of how I perceive OctoPrint's user base?

EmEiBee, 2023-08-31:
.. it's wrong that you accuse me of "passive aggressive" behavior. Assuming this is true; don't I have the same right to accuse you of a misperception of the user base?
It may be that we are both on the wrong track here, but mutual insinuations, above all the attempt to see something negative in every sentence without thinking about whether it simply means what is written, will not work help further, but seems to be kind of "woke" these days...
So, in order to wrap things up and not escalate, I think it's better to stick with my "passive aggressive" suggestion and just let things go...
Is that ok for you?

kantlivelong, 2023-08-31:
Nothing useful coming out of this closed ticket.

Either open a valid ticket per issue or don't open one at all.