OctoPrint/OctoPrint #4880: [Request] Remove `scripts` from repo

foosel, 2023-08-29:
### Is your feature request related to a problem? Please describe.

There's a bunch of init scripts for sysv and systemd located in the source tree under `scripts`. That stuff hasn't really been maintained over the years, and while the systemd sample is still totally fine, just like the outdated sysv one it still relies on an installation into a venv in a user's home when best practice would actually be to create a dedicated user and application folder somewhere under `/opt` or similar.

### Describe the solution you'd like

Get rid of this stuff and have deployment information rather live in the docs or if needed in a dedicated repo.

Does anyone see any issue with that? Am I overlooking problems for any of the third party installer scripts that might be utilizing these files (@paukstelis maybe)?

cp2004, 2023-08-29:
The install guides on the forums have the user `wget` these files to use them in their install - that could change, but wanted to point out that as a reference.

paukstelis, 2023-08-29:
No impact on octoprint_deploy.

foosel, 2023-11-20:
Adjusted the install guide and got rid of the files. Will be gone in 1.10.0.

commit bfd7199a (2023-11-20): 🔥 Get rid of unmaintained scripts folder

Closes #4880