OctoPrint/OctoPrint #4882: Default serverRestartCommand does not cleanly, quickly shutdown plugins

rich-dobbs-13440, 2023-08-30:
### The problem

1) I'm starting a background thread in on_after_startup, in implementing octoprint.plugin.StartupPlugin.
2) I intend to shutdown this thread in on_shutdown, in implementing  octoprint.plugin.ShutdownPlugin.
3) Based on lack of logging message, when the default serverRestartCommand is issued, on_shutdown is never called when  giving the serverRestartCommand, either by using the web page, a local Raspberry PI command, or a command via SSH. 
4) Without a clean shutdown of the background threads, the restart takes minutes, rather than the expected seconds.

I got around this problem for development by implementing a SimpleAPICommand to directly call my on_shutdown method, but that required using an API key.  But when I roll out my plugin, this issue will happen to end-users.  

Based on the Octoprint Community discussions, this is a long term, reoccuring problem, not specific to my code.  

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

 OctoPrint 1.9.2 Python 3.9.2 OctoPi 1.0.0

### Operating system running OctoPrint

Octopi - running on RasberryPI 4 Model B - 8 GB Ram

### Printer model & used firmware incl. version

Not applicable

### Browser and version of browser, operating system running browser

I'm testing using Brave browser  v1.57.53 from Linux Mint 21 Cinnamon

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230830112802.zip](https://github.com/OctoPrint/OctoPrint/files/12478589/octoprint-systeminfo-20230830112802.zip)

rich-dobbs-13440, 2023-08-30:
As additional background, I tried to get around this issue by implementing the octoprint.plugin.EventHandlerPlugin mixin, but the event never was signaled.  Having not used this method previously, I have no confidence in my implementation.  So here is what I wrote:

    def on_event(self, event, payload):
        if event == "Shutdown":
           self._logger.info("In on_event, with event == 'Shutdown'")
           self.on_shutdown()
           
I found no logger message when going through a restart, nor did implementing this code alter how long it took for the restart to happen.

rich-dobbs-13440, 2023-08-30:
FYI, the default command from config.yaml is:

server:
  commands:
    serverRestartCommand: sudo service octoprint restart

cp2004, 2023-08-30:
There isn't a default in OctoPrint, this only comes with preconfigured setups (and so OctoPi is included) - so it's not necessarily systemctl doing the work, but often this is the case.

I believe you'll need to set the `daemon` flag on your background threads to ensure OctoPrint can handle the shutdown & exit cleanly. Give that a go, and see what happens. It can either be done in the construction or afterwards on the thread object before it's started.

I'm not going to claim to be a true expert in threading usage, but I spent a long time on this problem - it particularly reared its head when multiprocessing became involved. If the thread/process could catch the shutdown SIGTERM signal, it may be able to exit itself as well. As non-daemon threads, I suspect something is waiting for the threads to stop *before* handling the shutdown (this is the bit I'm not certain about). But setting them to daemon means that we're not waiting for them to exit *before* OctoPrint shuts down.

In general, restarts of the service should be fine, but often on shutdown I've seen that the Pi does not hang around for long enough every time to let stuff be exited if it's taking a second to do so, and just shuts down.

rich-dobbs-13440, 2023-08-31:
Hi Charlie Powell,

Thanks for your response.  Since I'm targeting a plugin for use in Octopi,
for me at least, that is the default.  I'll check into the daemon flag, but
that seems like a workaround to me.  It seems to me that based on the
limited documentation, that I should be able to use the shutdown plugin
callback, which is never triggered.  What if there are specific things that
must happen upon shutdown, rather than just terminating a thread?  Maybe
because of the way that the default restart is implemented, folks have
blown away having a working shutdown plugin mixin, but if that is the case
then it should be adequately documented.

Best regards,

Rich Dobbs

foosel, 2023-11-28:
So, first of all, @cp2004 is right in that you need to set `daemon=True` on your thread if you want it to actually exit when its parent exits. See [the python docs on that](https://docs.python.org/3/library/threading.html?highlight=threading#threading.Thread.daemon).

That aside though, it looks like we have a behaviour change in that regard from from OctoPi < 1.0.0 to OctoPi >= 1.0.0.

Example:

OctoPi 0.18.0:

```
2023-11-28 17:52:47,473 - octoprint.server - INFO - Shutting down...
2023-11-28 17:52:47,485 - octoprint.server - INFO - Calling on_shutdown on plugins
2023-11-28 17:52:47,489 - octoprint.events - INFO - Processing shutdown event, this will be our last event
2023-11-28 17:52:47,672 - octoprint.events - INFO - Event loop shut down
2023-11-28 17:52:47,921 - octoprint.plugins.tracking - INFO - Sent tracking event shutdown, payload: {}
2023-11-28 17:52:48,111 - octoprint.server - INFO - Goodbye!
2023-11-28 17:52:55,660 - octoprint.startup - INFO - ******************************************************************************
2023-11-28 17:52:55,663 - octoprint.startup - INFO - Starting OctoPrint 1.9.3
2023-11-28 17:52:55,665 - octoprint.startup - INFO - ******************************************************************************
```

OctoPi 1.0:

```
2023-11-28 17:58:10,696 - octoprint.plugins.github_release_patcher - INFO - Started dummy release server on http://127.0.0.1:50672
2023-11-28 17:58:10,696 - octoprint.plugins.github_release_patcher - INFO - Set Github release URL to http://127.0.0.1:50672/releases/{user}/{repo}
2023-11-28 18:01:21,363 - octoprint.startup - INFO - ******************************************************************************
2023-11-28 18:01:21,366 - octoprint.startup - INFO - Starting OctoPrint 1.9.3
2023-11-28 18:01:21,367 - octoprint.startup - INFO - ******************************************************************************
```

No shutdown events triggered. It looks like the systemd unit on 1.0 simply nukes away the server directly, whereas on OctoPi < 1.0 it still got a chance to cleanly shut down. That's something that needs to get looked at, but that's rather something for OctoPrint/OctoPi-UpToDate or guysoft/OctoPi.

foosel, 2023-11-28:
For now this is tracked further in OctoPrint/OctoPi-UpToDate#13. Closing this since the code in general is there, it just doesn't get a chance to run by the runtime environment.

closed 2023-11-30 by foosel