OctoPrint/OctoPrint #4883: Plugin manager should disable the *Get More* button when printer is printing or paused

mitchscobell-availity, 2023-08-31:
### The problem

While printing I am still able to go into the Plugin manager and click the *Get More* button and search for and try and install plugins. This should probably be disabled like how the rest of the plugin manager functions. It doesn't appear to work and hangs which is probably good but instead we could just disable the button like the rest of the page so it can't be attempted!

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.2 

### Operating system running OctoPrint

Linux, octopi_version: 1.0.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20230831150307.zip](https://github.com/OctoPrint/OctoPrint/files/12489894/octoprint-systeminfo-20230831150307.zip)

jneilliii, 2023-08-31:
As of 1.9.0 queued plugin installs are possible, so if you click install it will queue those to complete after the print finishes.

mitchscobell-availity, 2023-08-31:
> As of 1.9.0 queued plugin installs are possible, so if you click install it will queue those to complete after the print finishes.

It didn't make that obvious, I couldn't close the install screen and had to refresh the page to use OctoPrint again. Maybe a message saying it was queued and the ability to close then?

cp2004, 2023-08-31:
That does sound like something wonky - it should work similar to that, at least that's the intention.

jneilliii, 2023-08-31:
Definitely not normal. I added screenshots to the PR that was merged that shows what it should do [here](https://github.com/OctoPrint/OctoPrint/pull/4643).

mitchscobell-availity, 2023-09-01:
Here's what mine is doing, it starts to install and just hangs here. I can't close the modal or do anything else on the page. If I refresh the page it keeps printing and it says it's queued.

[image]

jneilliii, 2023-09-01:
That's definitely unexpected. Will probably need to see what's in your browser's developer tools console when doing that install process.

mitchscobell-availity, 2023-09-01:
I just tested it on my other octoprint instance, and it worked exactly how you described. Funny enough the one I had issues with is a fresh image/install while the one that works I've had built for a few years and have been upgrading it as we go. Anything I can try and find in the dev tools to look at? I've reproduced this on two different computers, both using Chrome, one a Mac and one a Windows 10.

jneilliii, 2023-09-01:
There was no apparent error in your system info bundle showing a possible cause. In your browser press F12 to open developer tools and look at the console tab in there to see if any errors are showing during the process.

sbroweleit-insight, 2023-10-13:
Also here is a UI issue in that same area
[image]

foosel, 2023-10-17:
Cannot reproduce

[image]

Also, different thing altogether, open a new bug report please, *once made sure it's also happening in safe mode*.

foosel, 2023-11-28:
@mitchscobell-availity Can you still reproduce this, and if so please do what @jneilliii asked for? This is currently unreproduced, and so far only observed by you on one of your systems as far as we know.

closed 2024-01-08 by foosel