OctoPrint/OctoPrint #4892: Real time tool change

andraisr, 2023-10-06:
### Is your feature request related to a problem? Please describe.

No, this feature is not related to an issue.

### Describe the solution you'd like

Real-time tool change for printers with multi-extruders. When actively printing, in order to change to another extruder, ( for example running out of filament), Allow for simply selecting another tool. If printing is set for T0, allow for selecting T1 for continuing printing. Currently, I have to pause the print, then switch tool on the Control menu. then resume printing. This causes a blob on the print. Most all other control menus for slicers printing via usb do this already.

Currently that switch tool feature is inactive during printing.

### Describe alternatives you've considered

Printing direct from usb. not using Octoprint.

andraisr, 2023-10-06:
[octoprint-systeminfo-20231006151852.zip](https://github.com/OctoPrint/OctoPrint/files/12835758/octoprint-systeminfo-20231006151852.zip)

andraisr, 2023-10-06:
Thanks for your time in reviewing this.

foosel, 2024-04-25:
1.10.0 has been released.

closed 2024-04-25 by foosel