OctoPrint/OctoPrint #4894: [Request] Add API key QRCode to User Settings -> Application Keys

credomane, 2023-10-24:
### Is your feature request related to a problem? Please describe.

If you go to `Settings (wrench icon) -> Application Keys` then you can see all application keys for all users. You even have a little hourglass icon to click on to get details on a particular application key and even get a QRcode for it. This is great.

If you go to the `User Settings (person icon) -> Application Keys` then the logged in user sees only theor keys. This is also great. However there is only the delete button; the hourglass icon isn't there for details on that key and therefore no QRcode available. This is not so great.

### Describe the solution you'd like

The ability for a user to get a QRcode for their API Keys in their user settings.

### Describe alternatives you've considered

N/A

### Additional context

N/A

credomane, 2023-10-24:
Decided to dig into this myself and discovered that I might have this feature request in the wrong place?

I missed it before due to the screen resolution of the computer I was on but the main page of the user settings actually has a API key which shows a QRCode.  While the "Application Keys" tab seems to be provided by the "Application Keys Plugin" that is bundled with Octoprint.

foosel, 2023-11-22:
Merged the PR of @credomane (thanks!), this is now ready for 1.10.0.

commit 3fae2952 (2023-11-22): 🚸 Add APIKey details to user settings so QRCode can be viewed. (#4895)

Closes #4894