OctoPrint/OctoPrint #4901: [Request] Disable "ongoing print job will be canceled"

LoganDark, 2023-10-28:
### Is your feature request related to a problem? Please describe.

Klipper sometimes returns errors that are dumb.

### Describe the solution you'd like

Don't cancel the entire print job just because of a single "error".

### Describe alternatives you've considered

None.

### Additional context

[image]

This makes no sense and should not cancel a 4 hour print.

LoganDark, 2023-10-28:
Right now this can be mitigated by enabling relative extrusions in your slicer software (i.e. Cura)

that way, if you have to manually back the extruder out a bit during a print (I haven't calculated my esteps properly apparently...) it won't freak out

LoganDark, 2023-10-28:
[image]

lmAo I can't even have it move the print head up when I pause the print, because resuming the print will cause an error like this. F

cp2004, 2023-10-30:
This is configurable already:

[image]

It defaults to disconnecting the printer, so you must have changed it to cancelling the print job at some point.

Just moving the print head should not cause any extrusion, so I would probably recommend investigating the underlying issue, maybe you have some dodgy pause/resume scripts setup. You do have to get relative vs. absolute extrusion right in these scripts or your printer would *actually* try and retract 500mm instead of your intention. It's trying to protect you from causing damage.

closed 2023-10-30 by cp2004

LoganDark, 2023-10-30:
> Just moving the print head should not cause any extrusion

When the next instruction in the G-code is an extruding move, I think it will try to extrude on the way down from where you moved it, rather than moving back to where it is supposed to be first.

cp2004, 2023-10-30:
> When the next instruction in the G-code is an extruding move, I think it will try to extrude on the way down from where you moved it, rather than moving back to where it is supposed to be first.

That's not normally how gcode works, if I am understanding your comment right.

*(assuming absolute positioning)*
```gcode
G1 X10
G1 E1
```
Will first move X to 10, then after that is complete move (extrude) E to 1, while
```gcode
G1 X10 E1
```
Will move X to 10 and E to 1 simultaneously.

LoganDark, 2023-10-30:
> That's not normally how gcode works, if I am understanding your comment right.
>
> _(assuming absolute positioning)_
>
> ```gcode
> G1 X10
> G1 E1
> ```
>
> Will first move X to 10, then after that is complete [...]

Yes and that instruction doesn't specify the *source* of the move because it is assumed to be "where the head currently is", which is fine, except for when the head has *moved since the print was paused*.

cp2004, 2023-10-30:
Right, you're correct in that sense. OctoPrint doesn't add any extra commands beyond what you tell it to because there's a huge range of use cases and firmware variants, use of absolute vs. relative extrusion etc., that the gcode scripts are left to the user to setup to their liking.

You can find an example gcode scripts that will move the print head out of the way on pause and then put it back to place here:https://docs.octoprint.org/en/master/features/gcode_scripts.html#more-nifty-pause-and-resume