OctoPrint/OctoPrint #4902: [Discovery plugin] wrong IP address sent outin discovery response in case of multiple interfaces

dadosch, 2023-10-28:
### The problem

1. OctoPrint is available from **two different networks** 192.168.**178**.1/24 and 192.168.**179**.1/24.
2. On any program (in the **178** subnet) using the autodiscovery via zeroconf/avahi, wait for the discovery of OctoPrint.
3. Autodiscovery works, but a wrong response is returned by octoprint for this discover request: The host name (octopi.local) [correct] and the address 192.168.**179**.17 [wrong, it should send this response to requests from the .179 subnet, and not from .178 subnet; and a correct response to requests from the .178 subnet]
4. Alternately, one can browse the requests sent out using avahi browse. Also there, you can see the wrong response is sent  (to the PC in the 178 subnet)

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.3

### Operating system running OctoPrint

OctoPi 1.0.0

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

This is probably the same thing as in https://github.com/OctoPrint/OctoPrint/issues/4359 but the mentioned options ignoredAddresses and ignoredInterfaces don't apply in my humble opinion, since OctoPrint should be discoverable from both interfaces in this case, with the correct response in each interface context.

No specific interfaces or addresses are specified in OctoPrints config.

For the pictures: Note this is taken on a PC which is **not** in the 179 subnet, only in the 178 subnet.
Avahi browse results:
[image]
PrusaSlicer Discovery Result
[image]
Again, these screenshots are taken on a machine **WHICH IS NOT IN THE .179** subnet.

[octoprint-systeminfo-20231029001912.zip](https://github.com/OctoPrint/OctoPrint/files/13196918/octoprint-systeminfo-20231029001912.zip)

Lastly, I want to thank you very much for maintaing and putting so much love into OctoPrint :)

foosel, 2023-11-28:
That's sadly an issue with the underlying `zeroconf` library.

See https://github.com/python-zeroconf/python-zeroconf/issues/296#issuecomment-866188035, python-zeroconf/python-zeroconf#180 and python-zeroconf/python-zeroconf#806. 

I depend on a library here since I'd rather not reinvent the wheel, and I'm not familiar enough with the library code in question to lend a helping hand here, nor am sure would it even be welcome at the current point.

foosel, 2024-08-19:
Closing since this is out of my hands, see above.

closed 2024-08-19 by foosel