OctoPrint/OctoPrint #4904: ERROR - Exception on /api/login on unsupported flask (2.3.3) and werkzeug (2.3.7) version

gador, 2023-11-03:
### The problem

__This is just a head-up, not an actual bug!__

It seems octoprint is currently incompatible with newer `flask` versions.

This isn't a bug per se (as the working version of `flask` and `werkzeug` are pinned in octoprint), but rather something I came across while maintaining octoprint for `NixOS`. 
Due to how `NixOS` works, it usually always pulls in the newest versions of all packages. While it did so for octoprint, the following error appeared when trying to login:
```
Okt 31 18:35:33 kari octoprint[2265004]: 2023-10-31 18:35:33,201 - octoprint - ERROR - Exception on /api/login [POST]
Okt 31 18:35:33 kari octoprint[2265004]: Traceback (most recent call last):
Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/flask/app.py", line 2190, in wsgi_app          Okt 31 18:35:33 kari octoprint[2265004]:     response = self.full_dispatch_request()
Okt 31 18:35:33 kari octoprint[2265004]:                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/flask/app.py", line 1486, in full_dispatch_request
Okt 31 18:35:33 kari octoprint[2265004]:     rv = self.handle_user_exception(e)
Okt 31 18:35:33 kari octoprint[2265004]:          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/flask/app.py", line 1482, in full_dispatch_request
Okt 31 18:35:33 kari octoprint[2265004]:     rv = self.preprocess_request()                                                                                                                 Okt 31 18:35:33 kari octoprint[2265004]:          ^^^^^^^^^^^^^^^^^^^^^^^^^
Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/flask/app.py", line 1974, in preprocess_request
Okt 31 18:35:33 kari octoprint[2265004]:     rv = self.ensure_sync(before_func)()                                                                                                           Okt 31 18:35:33 kari octoprint[2265004]:          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/octoprint/server/__init__.py", line 1519, in before_request
Okt 31 18:35:33 kari octoprint[2265004]:     g.locale = self._get_locale()                                                                                                                  Okt 31 18:35:33 kari octoprint[2265004]:                ^^^^^^^^^^^^^^^^^^
Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/octoprint/server/__init__.py", line 1403, in _get_locale
Okt 31 18:35:33 kari octoprint[2265004]:     if "l10n" in request.values:
Okt 31 18:35:33 kari octoprint[2265004]:                  ^^^^^^^^^^^^^^                                                                                                                    Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/werkzeug/utils.py", line 106, in __get__
Okt 31 18:35:33 kari octoprint[2265004]:     value = self.fget(obj)  # type: ignore
Okt 31 18:35:33 kari octoprint[2265004]:             ^^^^^^^^^^^^^^
Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/werkzeug/wrappers/request.py", line 466, in values
Okt 31 18:35:33 kari octoprint[2265004]:     sources.append(self.form)
Okt 31 18:35:33 kari octoprint[2265004]:                    ^^^^^^^^^
Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/werkzeug/utils.py", line 106, in __get__                        Okt 31 18:35:33 kari octoprint[2265004]:     value = self.fget(obj)  # type: ignore
Okt 31 18:35:33 kari octoprint[2265004]:             ^^^^^^^^^^^^^^
Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/werkzeug/wrappers/request.py", line 446, in form                Okt 31 18:35:33 kari octoprint[2265004]:     self._load_form_data()                                                                                                                         Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/flask/wrappers.py", line 114, in _load_form_data
Okt 31 18:35:33 kari octoprint[2265004]:     super()._load_form_data()
Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/werkzeug/wrappers/request.py", line 271, in _load_form_data
Okt 31 18:35:33 kari octoprint[2265004]:     self._get_stream_for_parsing(),                                                                                                                Okt 31 18:35:33 kari octoprint[2265004]:     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/werkzeug/wrappers/request.py", line 298, in _get_stream_for_parsing
Okt 31 18:35:33 kari octoprint[2265004]:     return self.stream
Okt 31 18:35:33 kari octoprint[2265004]:            ^^^^^^^^^^^                                                                                                                             Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/werkzeug/utils.py", line 106, in __get__
Okt 31 18:35:33 kari octoprint[2265004]:     value = self.fget(obj)  # type: ignore
Okt 31 18:35:33 kari octoprint[2265004]:             ^^^^^^^^^^^^^^
Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/werkzeug/wrappers/request.py", line 350, in stream
Okt 31 18:35:33 kari octoprint[2265004]:     return get_input_stream(
Okt 31 18:35:33 kari octoprint[2265004]:            ^^^^^^^^^^^^^^^^^
Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/werkzeug/wsgi.py", line 175, in get_input_stream
Okt 31 18:35:33 kari octoprint[2265004]:     content_length = get_content_length(environ)
Okt 31 18:35:33 kari octoprint[2265004]:                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^
Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/werkzeug/wsgi.py", line 129, in get_content_length
Okt 31 18:35:33 kari octoprint[2265004]:     return _sansio_utils.get_content_length(
Okt 31 18:35:33 kari octoprint[2265004]:            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/werkzeug/sansio/utils.py", line 157, in get_content_length
Okt 31 18:35:33 kari octoprint[2265004]:     return max(0, _plain_int(http_content_length))
Okt 31 18:35:33 kari octoprint[2265004]:                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Okt 31 18:35:33 kari octoprint[2265004]:   File "/nix/store/jq5gma3a9g7wgzrf2sfrf9p0jn8giran-python3-3.11.5-env/lib/python3.11/site-packages/werkzeug/_internal.py", line 326, in _plain_int
Okt 31 18:35:33 kari octoprint[2265004]:     value = value.strip()
Okt 31 18:35:33 kari octoprint[2265004]:             ^^^^^^^^^^^
Okt 31 18:35:33 kari octoprint[2265004]: AttributeError: 'int' object has no attribute 'strip'
```

Downgrading to `flask` 2.2.5 and `werkzeug`  2.2.3 (last versions in `NixOS` before the jump to `flask`2.3) fixes the "issue". 

Issue on `NixOS`: https://github.com/NixOS/nixpkgs/pull/264650/

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### Version of OctoPrint

1.9.3

### Operating system running OctoPrint

aarch64-linux

### Printer model & used firmware incl. version

Ender 3

[... cut ...]
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

No Systeminfo Bundle info, as the fixed version already runs. Did not test in safe mode (didn't think of it at the time)

illuusio, 2023-11-09:
I've en courted this too and made small patch (which not **fix** but rather workaround) after applying patch everything works smooth with Flask 2.2.3 and Werkzeug 3.0.1 Tumbleweed distro packages:
```
--- werkzeug-3.0.1/src/werkzeug/_internal.py    2023-10-24 23:52:38.000000000 +0300
+++ werkzeug-3.0.1-patch/src/werkzeug/_internal.py      2023-11-09 09:40:47.123773929 +0200
@@ -207,8 +207,9 @@
 
     Any leading or trailing whitespace is stripped
     """
-    value = value.strip()
-    if _plain_int_re.fullmatch(value) is None:
-        raise ValueError
+    if isinstance(value, int) is False:
+       value = value.strip()
+       if _plain_int_re.fullmatch(value) is None:
+          raise ValueError
 
     return int(value)
```

Problem seems to be that method `get_content_length` returns int not string as it used to. This should work also on version which in report. I don't understand why this happening as it seems to be inner werkzeug thing or it's Python 3.11.

Probably should report on Werkzeug?

foosel, 2023-11-28:
That indeed explodes way outside of OctoPrint's code base, so that's definitely something that rather needs reporting upstream.

I regularly check compatibility against newer versions and schedule migrations as needed, but the version pins are there for a reason, and no proper functionality can be guaranteed when running versions of third party libraries outside of their defined ranges.

closed 2023-11-28 by foosel