OctoPrint/OctoPrint #4905: CGode viewer does not parse a particular GCode file in Firefox on Win10

JanKalin, 2023-11-10:
### The problem

After slicing an STL with PrusaSlicer v2.6.1-win64, uploading to OctoPrint, and printing it, I cannot view the GCode progress in Firefox. Version is 119.0.1 (latest). This is true for both the regular and private modes (in which almost all the plugins are disabled). The same file works OK in Edge and Brave browsers. This is the first time that I came across such a GCode file.

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

It is not an issue of the printing part of OctoPrint

### Version of OctoPrint

1.9.3

### Operating system running OctoPrint

OctoPi* 0.18.0 (build 2022.03.23.100241)

### Printer model & used firmware incl. version

Prusa i3 Mk3S

### Browser and version of browser, operating system running browser

Firefox 119.0.1, Windows 10 Pro 22H2

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20231110165938.zip](https://github.com/OctoPrint/OctoPrint/files/13321128/octoprint-systeminfo-20231110165938.zip)
[image]
[shell_0.6n_0.35mm_PETG_MK3S_5h19m.zip](https://github.com/OctoPrint/OctoPrint/files/13321152/shell_0.6n_0.35mm_PETG_MK3S_5h19m.zip)
[console_log.txt](https://github.com/OctoPrint/OctoPrint/files/13321170/console_log.txt)

foosel, 2023-11-10:
> It is not an issue of the printing part of OctoPrint

That doesn't matter, especially not in case of issues with the gcode viewer.

However, this very much looks like a duplicate of #4879 - a browserbug reported to Firefox that is sadly not yet fixed upstream.

closed 2023-11-10 by foosel

JanKalin, 2023-11-10:
Ah, ok. And yes, the file in question is large...