OctoPrint/OctoPrint #4906: logViewModel dependency incorrect

jneilliii, 2023-11-12:
### The problem

When developing a plugin it seems the docs call out a dependency name of `logViewModel` [here](https://docs.octoprint.org/en/master/plugins/viewmodels.html#dependencies). When attempting to use this in a plugin it errors out in console about dependency failure, but if I use logsViewModel as defined [here](https://github.com/jneilliii/OctoPrint-DebugEnabled/blob/bb4f61706de039d50d919a591ac16739c3d17022/octoprint_debug_enabled/static/js/debug_enabled.js#L40) and appears to be what is used in the logging plugin's additionalNames [here](https://github.com/OctoPrint/OctoPrint/blob/dfe289304943b9622605edb49e245a3f22d9ceae/src/octoprint/plugins/logging/static/js/logging.js#L399) it works without error. Not sure if this is a documentation error or an additionalNames error? 

### Did the issue persist even in safe mode?

I cannot test this issue in safe mode (state why below)

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

requires a plugin to be able to be loaded to demonstrate

### Version of OctoPrint

OctoPrint 2.0.0.dev2177+g82a65cdd3

### Operating system running OctoPrint

Windows

### Printer model & used firmware incl. version

N/A

### Browser and version of browser, operating system running browser

Chrome Version 119.0.6045.124 (Official Build) (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20231112002037.zip](https://github.com/OctoPrint/OctoPrint/files/13327257/octoprint-systeminfo-20231112002037.zip)

[localhost-1699766552924.log](https://github.com/OctoPrint/OctoPrint/files/13327259/localhost-1699766552924.log)

[image]

cp2004, 2023-11-20:
I'd say this should just be changed in the docs - changing it in the code brings the *potential* that someone is using it and it breaks, but if they followed the docs it's definitely broken. Regardless of which was the intention, it should be `logsViewModel`

foosel, 2023-11-21:
Agreed. Will push a docs change.

commit 3953f630 (2023-11-21): 📝 logViewModel -> logsViewModel

Closes #4906
closed 2024-04-24 by foosel