OctoPrint/OctoPrint #4907: Virtual Printer Communication Timeout (SerialTimeoutException)

bdwilson, 2023-11-12:
### The problem

1. Install OctoPi 1.0 with new camera stack and with OctoPrint 1.9.3
2. Enable Virtual Printer plugin, Obico and Octoslack
3. Connect to virtual printer. 
4. Print a dummy gcode file (choose one >1hr from here https://github.com/bdwilson/obico-bambu-octoprint/tree/main/g-code).
5. Virtual printer disconnects with the following errors; will attach logs and system info (Raspi Zero 2W).  BTW, timeouts in virtual.py appear to be hardcoded and do not appear to be adjustable from config.yml. 

```
| Recv: T:21.30/ 0.00 B:21.30/ 0.00 @:64
| Recv: Not SD printing
| Recv: ok
| Recv: Error: expected line 113 got 112
| Send: N115 M117 INDICATOR-Layer56*45
| Recv: Resend:113
2023-11-12 11:06:10,405 - octoprint.util.comm - INFO - Communication timeout during an active resend, resending same line again to trigger response from printer.
2023-11-12 11:06:20,419 - octoprint.plugins.virtual_printer.VirtualPrinter - INFO - Incoming queue is full, raising SerialTimeoutException
2023-11-12 11:06:30,422 - octoprint.plugins.virtual_printer.VirtualPrinter - INFO - Incoming queue is full, raising SerialTimeoutException
2023-11-12 11:06:30,423 - octoprint.util.comm - ERROR - Unexpected error while writing to serial port
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/virtual_printer/virtual.py", line 2053, in write
    written = self.incoming.put(
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/virtual_printer/virtual.py", line 2317, in put
    raise queue.Full
queue.Full

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/comm.py", line 4957, in _do_send_without_checksum
    result = self._serial.write(to_send)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/virtual_printer/virtual.py", line 2062, in write
    raise SerialTimeoutException()
serial.serialutil.SerialTimeoutException

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/virtual_printer/virtual.py", line 2053, in write
    written = self.incoming.put(
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/virtual_printer/virtual.py", line 2317, in put
    raise queue.Full
queue.Full

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/util/comm.py", line 4966, in _do_send_without_checksum
    result = self._serial.write(to_send)
  File "/home/pi/oprint/lib/python3.9/site-packages/octoprint/plugins/virtual_printer/virtual.py", line 2062, in write
    raise SerialTimeoutException()
serial.serialutil.SerialTimeoutException
2023-11-12 11:06:30,667 - octoprint.util.comm - INFO - Please see https://faq.octoprint.org/serialerror for possible reasons of this.
2023-11-12 11:06:30,748 - octoprint.util.comm - INFO - Changing monitoring state from "Printing" to "Offline after error"
```
[octoprint-systeminfo-20231112111021.zip](https://github.com/OctoPrint/OctoPrint/files/13328543/octoprint-systeminfo-2023111
[octoprint.log](https://github.com/OctoPrint/OctoPrint/files/13328547/octoprint.log)
[plugin_virtual_printer_serial.log](https://github.com/OctoPrint/OctoPrint/files/13328548/plugin_virtual_printer_serial.log)
2111021.zip)

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.3

### Operating system running OctoPrint

Fresh install of Octopi 1.0; Pi zero 2 w. 

### Printer model & used firmware incl. version

Virtual Printer

### Browser and version of browser, operating system running browser

Edge

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

foosel, 2023-11-28:
While I was not able to reproduce the issue, I'm pretty sure what happened based on the provided logs.

A `G4 S54` triggered a `wait` response, since right after sending the `ok` the host didn't send a new incoming line fast enough to prevent that from happening. As a result, combined with several simulated transmition errors right after, things get so much out of balance that at some point the incoming queue runs full and an exception gets triggered, killing the connection with a simulated SerialTimeout.

I've added a fix that will hopefully prevent such a `wait` from being triggered in the first place.

I've also made the simulated errors configurable, by now setting `plugins.virtual_printer.simulated_errors` to an empty array `[]` in config.yaml, e.g.

``` yaml
plugins:
  virtual_printer:
    simulated_errors: []
```

Will all be in 1.10.0

bdwilson, 2023-11-30:
Will test and leave feedback once included. Thank you!

foosel, 2024-04-25:
1.10.0 has been released.

closed 2024-04-25 by foosel