OctoPrint/OctoPrint #4908: git tag checksums unstable

selfisekai, 2023-11-18:
github-generated tarballs have a little bit of git metadata embeded in the file.

https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/_version.py#L22-L26

in the current form, it keeps changing at least once after a release - when a new commit is added to master,
making the tag no longer the HEAD.

```diff
--- 1/OctoPrint-1.9.2/src/octoprint/_version.py
+++ 2/OctoPrint-1.9.2/src/octoprint/_version.py
@@ -23,7 +23,7 @@
     # setup.py/versioneer.py will grep for the variable names, so they must
     # each be defined on a line of their own. _version.py will just call
     # get_keywords().
-    git_refnames = " (HEAD -> master, tag: 1.9.2)"
+    git_refnames = " (tag: 1.9.2)"
     git_full = "f027aaa7300f70b046525a49245b33c43dd198c1"
     keywords = {"refnames": git_refnames, "full": git_full}
     return keywords
```

this is an issue in distributions, often a recurring one:
- https://gitlab.alpinelinux.org/alpine/aports/-/merge_requests/49445
- https://github.com/NixOS/nixpkgs/issues/261934
- https://aur.archlinux.org/cgit/aur.git/commit/?h=octoprint&id=90a6edd35284fac8c1ff4e8baf07486b6f3d9593

solutions to consider:
1. release own tarballs (generated in github actions) via github releases
2. rework the way git metadata is attached (to keep it stable)
3. remove the git format mark at all, bump version in a file manually?

selfisekai, 2023-11-18:
system info not applicable to the issue

foosel, 2023-11-20:
You could work around this by fetching the latest version from PyPi instead of GitHub, which already gets the fixed packages uploaded on release, but I'll also see that I can add the release packages as separate downloads to the release.

foosel, 2023-11-20:
Gotcha. As I said, I'll see that the release artifacts that are created by the release test and build workflow are attached to the release going forward, I hope that'll solve this. I have no control of the source zip however, and I need the embedded git metadata in there for automatic versioning.

foosel, 2023-11-21:
With the above commit it assets should hopefully be attached to the release from 1.10.0rc1 forward. We'll see if the CI actually does what it's supposed to do then. If not, I'll adjust the workflow as needed. Steps looked good on a test repo however.

foosel, 2024-01-31:
Good news, the builds for 1.10.0rc1 just automatically got attached to its GitHub release, so that automation indeed looks like it's working 🎉

selfisekai, 2024-01-31:
it seems to work, a wheel gets generated, but then there's no tests included ("collected 0 items")

foosel, 2024-01-31:
Oh, I thought you just required the actual build distribution, not a full checkout with the `test` folder included. That was sadly unclear.

I'll have to investigate if I can provide that. Source and wheel distributions don't include it by default, and I'm not sure I can create a tarball with static version and that, but I'll see what I can do for rc2/full release.

Just to make sure I don't misunderstand things, could you please describe the involved workflow? I take it you run tests locally, the build and install stuff? And no metadata must change in the meantime?

foosel, 2024-02-01:
I've now adjusted the build to run `git archive` and add the resulting source archive to the files attached to the build and on releases also the GitHub release.

Such a file would look roughly like this: 

[OctoPrint-1.10.0rc2.dev10+g0d3f440e7.source.tar.gz](https://github.com/OctoPrint/OctoPrint/files/14127538/OctoPrint-1.10.0rc2.dev10%2Bg0d3f440e7.source.tar.gz)

Could you verify that this solves your issue?

Once created it would not get touched again, and I'll also publish a `sha512sums.txt` file listing the sums of the wheel, the sdist tarball and the source tarball at build time to allow to ensure that.

foosel, 2024-02-02:
> this is an uncompressed tar,

Ooops, correct. There was an error in the `git archive` call parameters, that's now fixed: 
[OctoPrint-1.10.0rc2.dev12+g7e43e98f9.source.tar.gz](https://github.com/OctoPrint/OctoPrint/files/14138071/OctoPrint-1.10.0rc2.dev12%2Bg7e43e98f9.source.tar.gz)

> sha512sums are generated automatically with `abuild checksum`, that's very routine for all packages and them being provided wouldn't change anything for me

I figured as you didn't ask for it, but it might help others however, so I figured it couldn't hurt putting them on there alongside as well.

commit fcc99bd2 (2023-11-21): 👷 Publish release artifacts as release assets

Closes #4908
closed 2024-04-24 by foosel