OctoPrint/OctoPrint #4911: Webcam preview in fullscreen and testmode display the vertical placed webcam view incorrect

dirkvranckaert, 2023-11-21:
### The problem

I have 2 webcams configured on my OctoPrint installation, twice the popular Logitec webcam C270.
One is installed horizontal (as you usually use these webcams). One however (due to limitations of my enclosure) is installed vertically. And I'm pretty happy with that installation direction since that gives me quite a different look at my printer :-)

But the problem is that on two places the webcam preview is not correctly displayed:
1. As you can see in the screenshot below I have rotated my camera through the settings with 90°, however if I press the preview button it does not rotate.
[image]
[image]
2. The regular preview inline is going good
[image]
3. When I click the inline-preview for the fullscreen-view however it goes wrong, it does not scale the view to be fullscreen but instead the camera preview falls almost half off the screen...
[image]
een view

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.3

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Prusa MK4

### Browser and version of browser, operating system running browser

Chrome

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20231121194555.zip](https://github.com/OctoPrint/OctoPrint/files/13429954/octoprint-systeminfo-20231121194555.zip)

cp2004, 2023-11-21:
The option to watch the webcam in full screen is not part of OctoPrint core. Did you try it in safe mode?

dirkvranckaert, 2023-11-22:
Safemode does not make any difference. Ok for the fullscreen view (I'll try to find out who's plugin that is then) but still the previews in the settings are not reflecting my settings.

foosel, 2023-11-22:
The idea of the test preview is actually JUST to check whether the URL is correct, by embedding it directly with any kind of further processing. That's also why it's located next to the URL entry and not close to the rotation and flip settings. Changing that would basically require to duplicate the code already used in the control tab, which I'd rather not, and also make debugging more problematic - is the camera distorted due to a bug in the rotation/flip code, or because something's up with the camera? 

So this is working as it should here, and as already mentioned, the fullscreen view is not part of OctoPrint's core. That SHOULD probably react to the rotation and flip settings, however that needs to get reported to the plugin that provides that (maybe [Fullscreen Webcam](https://plugins.octoprint.org/plugins/fullscreen/)?).

As thus, I'm closing this since I don't see anything actionable here.

closed 2023-11-22 by foosel