OctoPrint/OctoPrint #4913: [Request]

pmzandbergen, 2023-11-30:
### Is your feature request related to a problem? Please describe.

If I connect Octoprimnt to the new Snapmaker Artisan no bed temperature is shown.

Note: Originally requested in https://github.com/OctoPrint/OctoPrint/issues/4746 (closed)

### Describe the solution you'd like

The Snapmaker Artisan delivers two bed temperatures. One for the inner zone and one for the outer zone. It would be great if Octoprint could handle this new feature with two temperture zones.

### Additional context

```
Recv:  T0:210.90 /210.00 B0:59.95 /60.00 B1:36.43 /0.00 M:44.26 /0.00 T0:210.90 /210.00 T1:37.70 /0.00 @:0 B0@:127 B1@:0 @0:0 @1:0 @B_ZONE: CENTER
Recv:  T0:210.40 /210.00 B0:59.92 /60.00 B1:36.53 /0.00 M:44.27 /0.00 T0:210.40 /210.00 T1:37.70 /0.00 @:127 B0@:127 B1@:0 @0:127 @1:0 @B_ZONE: CENTER
Recv:  T0:209.60 /210.00 B0:60.00 /60.00 B1:36.39 /0.00 M:44.43 /0.00 T0:209.60 /210.00 T1:37.70 /0.00 @:127 B0@:0 B1@:0 @0:127 @1:0 @B_ZONE: CENTER
Recv:  T0:209.10 /210.00 B0:59.90 /60.00 B1:36.43 /0.00 M:44.38 /0.00 T0:209.10 /210.00 T1:37.70 /0.00 @:127 B0@:127 B1@:0 @0:127 @1:0 @B_ZONE: CENTER
Recv:  T0:210.10 /210.00 B0:60.00 /60.00 B1:36.43 /0.00 M:44.36 /0.00 T0:210.10 /210.00 T1:37.70 /0.00 @:0 B0@:0 B1@:0 @0:0 @1:0 @B_ZONE: CENTER
```

The interesting part is: `B0:59.95 /60.00 B1:36.43 /0.00`

B0 = inner bed (in this example set to heat to 60 degree), B1 = outer bed (in this example not set to heat)

jneilliii, 2023-11-30:
This is related to how the temperature is being reported. If you were to install my [PlotlyTempGraph ](https://plugins.octoprint.org/plugins/plotlytempgraph/) plugin from the plugin repo, you'll see everything graphed that shows on that temperature report line and have options for what you want to turn on and off.

jneilliii, 2023-11-30:
There would still be work needed to support multiple beds for controlling the temperature from OctoPrint. In the interim you could use Custom Controls to add your own inputs on the Control tab for setting the different bed temps.

CDank, 2024-12-01:
Is there any solution for the bed temperatures on the Snapmaker Artisan?

faberc, 2024-12-04:
The only solution (and not really a solution) is to use the "Top Temps" plugin and add custom tools that have regexes that search the temp string for B0 and B1.

I don't know how this isn't an easy regex fix for the temp string in Octoprint. Something like 
```
B|B0:([^ ]+)
```

jneilliii, 2024-12-04:
The js side is here. https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/static/js/app/viewmodels/temperature.js

Felix2M, 2025-09-03:
I got the same problem and it is true PlotlyTempGraph can help with the visualization, but I want to use the bed temperature for my home assistant integration. Is there a possbility to do that? For me it would work if I could get just the inner bed temperature B0.

GitKassi, 2026-09-06:
It seems that this is not in focus at all - badly...

Actually this affects all bigger 3D printer as they normally have the inner and outer heater zone.
The Snapmaker Artisan is one of them with 400x400x400mm of working area.
I like OctoPrint very much but this is hopefully a feature which helps a lot more people.
The mentioned plugin shows both temperature and theire settings but only in the graph which is suboptimal.
I don't see the temperature in OctoPrint and can't control the value of the setting point(s). The bed temperature is showing as "-".

Is there anything you need to help in investigating for bringing this feature to a starting point?

foosel, 2026-09-07:
For being able to set the temperature it would be necessary to know HOW it can be set. Neither `M140` nor `M190` are documented to take a parameter defining the bed subarea. So that needs to be clarified first.

Furthermore, it would be necessary to know whether there is some way of detecting if the printer has multiple print bed zones during initial handshake (ideally a capability report), as there are plenty of weird firmware variants out there that might use `B0` and the like without it meaning that. 

If all you need is get `B0` reported as `B`, a single file plugin could help here that rewrites the returned temperature response's `B0:...` to `B:...`. Something [like this](https://gist.github.com/foosel/2e929b8221f5c34754460ad113b3e825) might work (NOT TESTED!).

Overall, for this to find its way into OctoPrint core, this needs to be more than just "some vendor firmwares do this undocumented thing".

mosswild, 2026-09-07:
> For being able to set the temperature it would be necessary to know HOW it can be set. Neither `M140` nor `M190` are documented to take a parameter defining the bed subarea. So that needs to be [...]

Both the modified M140 and M190 are documented in SnapMakers GCODE Artisan reference here: https://wiki.snapmaker.com/en/Snapmaker_Luban/manual/artisan_supported_gcode_references

They do take a parameters to target the central zone or the full zone.

jneilliii, 2026-09-07:
But the capability itself of supporting multiple bed zones is not reported in the capabilities report returned by the M115 command in the same docs?

Felix2M, 2026-09-07:
If there is no standardized way, it would be really appreciated if we could configure it by ourselves.

mosswild, 2026-09-07:
Yeah the capability isn't listed even when I run M115 on my machine. This could be a ticket to Snapmaker:

Send: N1 M115*39
Recv: FIRMWARE_NAME:Marlin V2.3.3-1210 (Dec 10 2025 10:42:55) SOURCE_CODE_URL:github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:STM32F407VGT6 EXTRUDER_COUNT:2 AXIS_COUNT:5 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
Recv: Cap:SERIAL_XON_XOFF:0
Recv: Cap:BINARY_FILE_TRANSFER:0
Recv: Cap:EEPROM:1
Recv: Cap:VOLUMETRIC:1
Recv: Cap:AUTOREPORT_POS:0
Recv: Cap:AUTOREPORT_TEMP:1
Recv: Cap:PROGRESS:0
Recv: Cap:PRINT_JOB:1
Recv: Cap:AUTOLEVEL:1
Recv: Cap:RUNOUT:1
Recv: Cap:Z_PROBE:1
Recv: Cap:LEVELING_DATA:1
Recv: Cap:BUILD_PERCENT:0
Recv: Cap:SOFTWARE_POWER:0
Recv: Cap:TOGGLE_LIGHTS:0
Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
Recv: Cap:EMERGENCY_PARSER:0
Recv: Cap:HOST_ACTION_COMMANDS:0
Recv: Cap:PROMPT_SUPPORT:0
Recv: Cap:SDCARD:0
Recv: Cap:REPEAT:0
Recv: Cap:SD_WRITE:0
Recv: Cap:AUTOREPORT_SD_STATUS:0
Recv: Cap:LONG_FILENAME:0
Recv: Cap:EXTENDED_M20:0
Recv: Cap:THERMAL_PROTECTION:1
Recv: Cap:MOTION_MODES:0
Recv: Cap:ARCS:1
Recv: Cap:BABYSTEPPING:1
Recv: Cap:CHAMBER_TEMPERATURE:1
Recv: Cap:COOLER_TEMPERATURE:0
Recv: Cap:MEATPACK:0
Recv: Cap:CONFIG_EXPORT:0
Recv: ok
Send: M21
Recv: echo:Unknown command: "M21"
Recv: ok
Send: M155 S2
Recv: ok
Recv:  T0:25.80 /0.00 B0:23.50 /0.00 B1:23.44 /0.00 M:33.72 /0.00 T0:25.80 /0.00 T1:26.20 /0.00 @:0 B0@:0 B1@:0 @0:0 @1:0 @B_ZONE: GLOBAL

Left that status string in to show how Artisan prints the zone temps and zone mode.

jneilliii, 2026-09-07:
I could see this being configured manually similar to how number of extruders in the printer profile is done. I think the trick would be how each of the various firmware variants handle the command for setting each zone. That's why I suggested creating your own custom control for that, as you can configure the command being sent. Unfortunately, it puts the commands on the Control tab and not the temp tab. 

[image]

[image]