OctoPrint/OctoPrint #4914: [Request] Add support for higher resolution displays

barkermn01, 2023-12-01:
### Is your feature request related to a problem? Please describe.

When using a 1440p screen the octopi does feel the right size it's also harder to use

### Describe the solution you'd like

implement the following CSS into the CSS
```css
@media screen and (min-width: 1400px) {
    
    .container.octoprint-container {
        width: 1280px;
    }
    
    .tabbable.span8 {width: 940px;}
    
    .jog-panel {
        margin-right: 10%;
        width:40%;
        
    }
    
    .jog-panel#control-jog-extrusion{
        margin-right: 10%;
        width:15%
    }
    
    .jog-panel#control-jog-general{
        margin-right: 10%;
        width:15%
    }
    
    div#control-jog-extrusion {
        margin: auto;
    }
    
    div#control-jog-general {
        margin: auto;
    }
    
    button#control-yinc{
        margin-left: 25px;
    }
    
    button#control-xdec.pull-left {
        margin-left: 25px;
    }
    
    button#control-ydec {
        margin-left: 25px;
    }
}
```

### Describe alternatives you've considered

not applicable

### Additional context

i have built this using the web inspector on my browser so i know it's good

cp2004, 2023-12-27:
I think there's quite a lot more CSS that would need to be edited to make it properly responsive. The [UI Customizer plugin](https://plugins.octoprint.org/plugins/uicustomizer/) provides some nice options for responsiveness in the UI.

While I would like to explore the options of some small CSS changes to make the UI less static, using similar tactics as the plugin does, we'd want to make sure the changes aren't too damaging for plugins. I also don't have time for that myself, it's just a dream.

closed 2023-12-27 by cp2004

jneilliii, 2023-12-27:
you could also implement this fix with [custom css](https://plugins.octoprint.org/plugins/customcss/) plugin?

barkermn01, 2023-12-28:
> you could also implement this fix with [custom css](https://plugins.octoprint.org/plugins/customcss/) plugin?

you could but and it will work as a work around for now. however, it should not be done via a plugin injecting custom CSS in for something that only adds support for sizing for a standard resolution setup, 1440p even chrome defines it as a laptop size which means this is well past being a small few people.