OctoPrint/OctoPrint #4916: Setting log format doesn't work.

phucly01, 2023-12-10:
### The problem

According to the document (https://mkdocs.octoprint.org/user-guide/configuration/logging-yaml/) , logging.yaml file should be in .octoprint directory on Linux where customized settting can be made.  However adding this to the yaml doesn't seem to have any affect (line number didn't show up in the octoprint.log):

loggers:
  octoprint:
    format: '%(asctime)s - %(name)s - %(levelname)s:%(lineno)d - %(message)s'
    level: INFO

Note:  the format line was added manually via vi but the rest were set by the gui

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.3

### Operating system running OctoPrint

Raspberry PI

### Printer model & used firmware incl. version

NA

### Browser and version of browser, operating system running browser

NA

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Not sure what Systeminfo Bundle is, pehaps more descriptive or instructive info is needed there.

2023-12-10 18:24:02,740 - octoprint.plugins.softwareupdate - INFO - Fetching check overlays from https://plugins.octoprint.org/update_check_overlay.json
2023-12-10 18:24:03,079 - octoprint.plugins.pluginmanager - INFO - Loaded notice data from disk, was still valid
2023-12-10 18:24:05,481 - octoprint.util.pip - INFO - Using "/home/phuc/oprint/bin/python -m pip" as command to invoke pip
2023-12-10 18:24:05,484 - octoprint.util.pip - INFO - pip installs to /home/phuc/oprint/lib/python3.9/site-packages (writable -> yes), --user flag needed -> no, virtual env -> yes
2023-12-10 18:24:05,486 - octoprint.util.pip - INFO - ==> pip ok -> yes
2023-12-10 18:24:05,533 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2023-12-10 18:24:05,706 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'base_url': 'http://simplypi.local/', 'path': '/', 'query_string': 'l10n=en'}
2023-12-10 18:24:20,670 - octoprint.server.preemptive_cache - INFO - ... done in 14.96s

phucly01, 2023-12-10:
[octoprint-systeminfo-20231210182847.zip](https://github.com/OctoPrint/OctoPrint/files/13629581/octoprint-systeminfo-20231210182847.zip)

jneilliii, 2023-12-11:
Seems to me you didn't follow the format of the yaml file as documented. If I'm understanding properly you are attempting to change the format, but the yaml file is expecting a `formatter` property and format section.

```
loggers:
  octoprint:
    formatter: simple
    level: INFO
formatters:
  simple:
    format: "%(asctime)s - %(name)s - %(levelname)s:%(lineno)d - %(message)s"
```

foosel, 2023-12-11:
As @jneilliii already pointed out, this is not how you change the formatting. You'll need to configured the format on the formatter, not the logger. This is just [how Python's logging framework works](https://docs.python.org/3/library/logging.config.html#logging-config-dictschema) which is being used here directly. There's also an example [in the docs](https://docs.octoprint.org/en/master/configuration/logging_yaml.html#changing-logging-formatters). However, I just noticed that this is a tad outdated if you also want to adjust the format used for the console output, as that usually now uses the `colored` formatter (unless instructed otherwise). So, to get what you want in all cases, you'd need to add this to your `logging.yaml`:

``` yaml
formatters:
  simple:
    format: "%(asctime)s - %(name)s - %(levelname)s:%(lineno)d - %(message)s"
  colored:
    format: "%(log_color)s%(asctime)s - %(name)s - %(levelname)s:%(lineno)d - %(message)s%(reset)s"
```

I've just adjusted the `maintenance` docs accordingly and those should be updated in a few minutes here: https://docs.octoprint.org/en/maintenance/configuration/logging_yaml.html

---

> Not sure what Systeminfo Bundle is, pehaps more descriptive or instructive info is needed there.

🤔 There **IS** a big section "Logs and other files needed for analysis" in the issue template - which the upload field and the checkbox are part of - with explanations and links:

[image]

foosel, 2023-12-11:
> According to the document (https://mkdocs.octoprint.org/user-guide/configuration/logging-yaml/)

I also just noticed this.... that's not the location of the official documentation, that's a WIP of a documentation rewrite. The docs are at docs.octoprint.org.

closed 2023-12-11 by foosel