OctoPrint/OctoPrint #4922: "Move or Rename File" - Directories not sorted

hakan42, 2023-12-23:
### The problem

The dropdown for "Move or Rename File" is not sorted in any discernible order. I already tried the Options (the wrench to the right) to sort files by Name or date but this impacts only the elements shown in the printable list but not the dropdown. Please see attached screenshot:

[image]

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.93.

### Operating system running OctoPrint

Linux, OctoPi 1.0.0

### Printer model & used firmware incl. version

not applicable, but Prusa MK3s

### Browser and version of browser, operating system running browser

Windows 11 Version 22H2 (Build 22621.2861), 120.0.6099.110 (Official Build) (64-bit) (cohort: Stable) 

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20231223122624.zip](https://github.com/OctoPrint/OctoPrint/files/13759033/octoprint-systeminfo-20231223122624.zip)

cp2004, 2023-12-27:
I would say the correct solution here would be to sort the folders in the same order that the normal file list is sorted in. I don't think there's a need to add another option within this dialog. Agreed?

hakan42, 2023-12-28:
yep, whatever order, it just needs to be predictable.

foosel, 2024-01-08:
The folder list was indeed not sorted at all. That was a quick fix that's now pushed (see above) and will be in 1.10.0.

commit aa027504 (2024-01-08): 🐛 Make sure to sort folder list

Closes #4922