OctoPrint/OctoPrint #4925: Newest update

Corocc, 2023-12-27:
### The problem

Newest update has ruined my printer
I have a savol sv06 plus, was working fine prior to update,it will print but the quality is shocking, try to do an xyz cube and right side basically looks like its trying to do a pyramid.
Have leveled and reset bed several times, reflashed my pi, even given the machine a deep clean.. basically pulled it apart 
Did a print on a flash card worked fine and came out amazing but second i pluged octo back it, back to problems 

Same problem after a reflash tells me yous did something in the code

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.3

### Operating system running OctoPrint

Octopi

### Printer model & used firmware incl. version

Savol sv06 plus on latest firmware

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[image]
[image]
[image]

Corocc, 2023-12-27:
No that is a security risk
Have provided you sufficient details for something visible to anyone
Open a direct line of chat please

jneilliii, 2023-12-27:
OctoPrint only sends the commands in the gcode to your printer. 

> No that is a security risk
don'ts see how, you can edit the files within the zip prior to uploading to redact any personal information.

>Have provided you sufficient details for something visible to anyone
not without the logs we can't tell what's going on. would recommend enabling serial.log in this situation as well to verify the commands being sent to your printer.

> Open a direct line of chat please
feel free to jump on the [discord](https://discord.octoprint.org) server, there is a support-printer channel just for this purpose.

closed 2023-12-27 by Corocc