OctoPrint/OctoPrint #4926: [Request] Report Layer# In Octoprint UI

tdunbar, 2024-01-08:
### Is your feature request related to a problem? Please describe.

If possible, could you add a feature to display the current layer number currently being printed.  I could really use this feature as I have settings in my slicer that only take effect upon reaching a specific layer.  I would like to observe my print at the moments just before the settings take effect.

### Describe the solution you'd like

I would like to see in realtime the current layer number being printed.  A good place would be in the State menu item on the main screen.

### Describe alternatives you've considered

The alternatives has been to watch my print for the settings to take effect.  This has not been practical since my prints take hours to complete.

foosel, 2024-01-08:
There's a plugin for that: https://plugins.octoprint.org/plugins/layerdisplay/

> I would like to see in realtime

Be aware that 100% realtime is not possible thanks to various buffers on the printer's side and no information at all from it back which commands are currently actually being running.

closed 2024-01-08 by foosel

cp2004, 2024-01-08:
There's also https://plugins.octoprint.org/plugins/DisplayLayerProgress/, and the GCode Viewer in OctoPrint reports the current layer if you select 'sync with job progress'.