OctoPrint/OctoPrint #4927: ValueError: could not convert string to float: '' - apply_temperature_offsets     temperature = float(groups["temperature"])

tr1plus, 2024-01-11:
### The problem

Sometimes octoprint crashes just before starting the purge line (after the heat up is done). I have to manually reconnect to the printer and re-start the print. Octoprint does save a recovery file, but I didn't use it yet due to the fact that the print never started so recovery wasn't required.

This happened 2 times now. Since it's my first time using octoprint (since yesterday), I don't have much experience beyond that. I'm also using the smart Preheat plugin, but I'm not sure how to tell where the crash/code below is coming from.

### Did the issue persist even in safe mode?

Yes, it did persist

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

it doesn't always crash in "normal" mode, so no need to go to safe mode

### Version of OctoPrint

1.9.3

### Operating system running OctoPrint

debian 12 VM on proxmox

### Printer model & used firmware incl. version

Ender 3 V3 SE

### Browser and version of browser, operating system running browser

chrome, windows or chromium, Linux mint

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240111205457.zip](https://github.com/OctoPrint/OctoPrint/files/13907711/octoprint-systeminfo-20240111205457.zip)

```
2024-01-11 20:42:59,955 - octoprint.util.comm - INFO - Changing monitoring state                                                                                                                                                                              from "Starting" to "Printing"
2024-01-11 20:44:02,280 - octoprint.util.comm - ERROR - Exception while processi                                                                                                                                                                             ng line
Traceback (most recent call last):
  File "/home/triplus/OctoPrint/lib/python3.11/site-packages/octoprint/util/comm                                                                                                                                                                             .py", line 5882, in getNext
    processed = self._process(line, offsets, current_tool)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/triplus/OctoPrint/lib/python3.11/site-packages/octoprint/util/comm                                                                                                                                                                             .py", line 5891, in _process
    return process_gcode_line(line, offsets=offsets, current_tool=current_tool)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/triplus/OctoPrint/lib/python3.11/site-packages/octoprint/util/comm                                                                                                                                                                             .py", line 6181, in process_gcode_line
    line = apply_temperature_offsets(line, offsets, current_tool=current_tool)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/triplus/OctoPrint/lib/python3.11/site-packages/octoprint/util/comm                                                                                                                                                                             .py", line 6149, in apply_temperature_offsets
    temperature = float(groups["temperature"])
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: could not convert string to float: ''
2024-01-11 20:44:02,281 - octoprint.util.comm - ERROR - Something crashed inside                                                                                                                                                                              the serial connection loop, please report this in OctoPrint's bug tracker:
Traceback (most recent call last):
  File "/home/triplus/OctoPrint/lib/python3.11/site-packages/octoprint/util/comm                                                                                                                                                                             .py", line 2566, in _monitor
    self._handle_ok()
  File "/home/triplus/OctoPrint/lib/python3.11/site-packages/octoprint/util/comm                                                                                                                                                                             .py", line 3296, in _handle_ok
    self._continue_sending()
  File "/home/triplus/OctoPrint/lib/python3.11/site-packages/octoprint/util/comm                                                                                                                                                                             .py", line 3517, in _continue_sending
    elif job_active and self._send_from_job():
                        ^^^^^^^^^^^^^^^^^^^^^
  File "/home/triplus/OctoPrint/lib/python3.11/site-packages/octoprint/util/comm                                                                                                                                                                             .py", line 4176, in _send_from_job
    line, pos, lineno = self._get_next_from_job()
                        ^^^^^^^^^^^^^^^^^^^^^^^^^
[... 13 lines cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/triplus/OctoPrint/lib/python3.11/site-packages/octoprint/util/comm                                                                                                                                                                             .py", line 6149, in apply_temperature_offsets
    temperature = float(groups["temperature"])
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ValueError: could not convert string to float: ''
2024-01-11 20:44:02,291 - octoprint.util.comm - INFO - Changing monitoring state                                                                                                                                                                              from "Printing" to "Offline after error"
2024-01-11 20:44:02,295 - octoprint.server - INFO - Starting autorefresh of seri                                                                                                                                                                             al port list
2024-01-11 20:44:02,296 - octoprint.plugins.action_command_notification - INFO -                                                                                                                                                                              Notifications cleared
2024-01-11 20:44:02,307 - octoprint.plugins.powerfailure - INFO - PowerFailure:                                                                                                                                                                              Print failed with error
2024-01-11 20:44:02,447 - octoprint.plugins.tracking - INFO - Sent tracking even                                                                                                                                                                             t commerror_crash, payload: {'commerror_text': 'See octoprint.log for details'}
2024-01-11 20:44:02,605 - octoprint.plugins.tracking - INFO - Sent tracking even                                                                                                                                                                             t print_failed, payload: {'origin': 'local', 'file': '77ceecb60813158935cf1e8c4a                                                                                                                                                                             5b2c5a0400f94d', 'elapsed': 376, 'reason': 'error', 'commerror_text': 'See octop                                                                                                                                                                             rint.log for details'}

```

[CE3V3SE_used-frame.zip](https://github.com/OctoPrint/OctoPrint/files/13907723/CE3V3SE_used-frame.zip)

Also using the following GCODE scripts as "before print job starts":

```
G28; auto home
G29; auto level

; Preheat
{% snippet 'doSmartPreheat' %}
```

cp2004, 2024-01-13:
Possibly the same as #4903?

Do you use the temperature offsets?

cp2004, 2024-01-13:
Almost certainly the same as the above issue.

Your gcode file contains this line (L23)
```
M109 S[material_print_temperature_layer_0]
```

Which matches the cause of #4903 

There is something wrong with your slicer that means it is not replacing these placeholders with the correct values. Perhaps this is in your slicer's start gcode and there is a typo or syntax issue from what it should be.

I'll close this as it's a duplicate of #4903, which is fixed for OctoPrint 1.10, but feel free to still follow up if you have questions/to let us know removing that line from your gcode fixed the issue.

closed 2024-01-13 by cp2004

tr1plus, 2024-01-13:
Thanks. I think I solved it based on your explanation.
I copied parts of the gcode from creality slicer to the print settings in cura to "fix" the location of the purge line. It seems creality slicer uses [] for variables and cura uses {}.

Apologies for logging the bug here - thanks for the great piece of software made :)

tr1plus, 2024-01-13:
Just started a print with the updated "variable" syntax. All working well. Thanks again!