OctoPrint/OctoPrint #4931: [Request] Add Digest authentication support for Classic Webcam

yhi264, 2024-01-14:
### Is your feature request related to a problem? Please describe.

Hi!

I used IP webcam(IO-DATA TS-WRLA) through Classic Webcam plugin. 
Stream URL:
http://user:pass@IP:port/snapshot.cgi
Snapshot URL:
http://user:pass@IP:port/snapshot.jpg

Stream URL works fine but Snapshot URL doesn't work.
I can access both URLs through web browser on my PC and wget from octopi SSL command line. And on curl I need digest auth option(--digest). 

API document of my IP webcam said that only support digest auth.
I think the only solution to use my IP webcam is to add Digest authentication support for Classic Webcam.

### Describe the solution you'd like

Please add Digest authentication support for Classic Webcam.

cp2004, 2024-05-22:
Fixed via https://github.com/OctoPrint/OctoPrint/commit/8437dc5aa24842f7d506f7f11af16213b8264959, for 1.11.0

foosel, 2025-04-22:
Closing this as 1.11.0 has just been released

closed 2025-04-22 by foosel