OctoPrint/OctoPrint #4932: [Request] Two-factor authentication (2FA/TFA)

JaneX8, 2024-01-15:
### Is your feature request related to a problem? Please describe.

The authentication mechanism does not support a form of multi factor authentication (MFA), two factor authentication (2FA/TFA).

### Describe the solution you'd like

I would really like to setup standard TOTP (or HOTP) method for two-factor authentication. I think there are plenty of suitable libraries to help with this including in Python and Javascript.

### Describe alternatives you've considered

Setting it up using PAM, similar to how I use 2FA on my SSH connections (https://github.com/google/google-authenticator-libpam). Which was already suggested here: https://github.com/OctoPrint/OctoPrint/issues/1936. However simply implementing TOTP is much simpler and easier than implementing PAM or OS-specific authentication.

foosel, 2024-01-22:
https://pyauth.github.io/pyotp/ looks like a good option for that

JaneX8, 2024-01-22:
It does indeed 😃

jneilliii, 2024-01-22:
> A nice to have would be in a multiuser environment of Octo to be able to enforce it for all users, so they are required to enable it at first or next login.

OctoPrint already supports multiple users. Check out the access control settings.

JaneX8, 2024-01-23:
I clarified that last 'nice to have'-sentence. Sounds good, thank you for your great work @foosel! _Hardware token support (browsers now support FIDO2) as a next feature request? 😏._

Don't forget to support @foosel on Patreon or other platforms.

foosel, 2024-01-23:
> Hardware token support (browsers now support FIDO2) as a next feature request? 😏.

I think I'll rather look into requiring exposing additional factors via some plugin interface and leave that to the community 😅 TOTP should already cover the bases for the majority of users.

jneilliii, 2024-01-23:
https://github.com/JoveToo/OctoPrint-MFA

JaneX8, 2024-01-23:
A few more thoughts about this that might be helpful for future implementation. I see them commonly weakly implemented:
- to avoid the risk of shoulder surfing, the input field for the OTP should be considered `type=password` and not `type=text` (hence invisible when being typed)
- (optionally) the OTP can be auto submitted after the last digit it typed, this reduced the time between the last user input and the actual submit, which decreases the timeframe for misusing the OTP, for example when captured using a keylogger
- the OTP should be invalidated after being used once. It's called a one-time-password for a reason. Often I see a simple validation of the OTP, which can be (re)used 100x as long as it is used within the right timeframe. This can for example be prevented through a simple serverside temporary caching mechanism. It prevents the risks of replay attacks where the code is reused directly after for example it being captured using a MitM attack.
- after an invalid OTP has been entered, one should be redirected to the first login step (username and password), this reduced the risk of brute-forcing the OTP code, note that a simple form of rate limiting should perhaps exist anyway on the regular login.
- a common practice is to allow the previous X and next X OTP's to be valid as well, to avoid time synchronization issues between the server and client. The secure practice is to keep both in sync and only allow the current code. But since that's likely inviable for this project, the amount of previous and next OTP's that are allowed should at least be somewhat limited or configurable. Projects like OctoPi could additionally for example make sure that time is synchronized serverside (for example ntp/chrony).

JaneX8, 2024-01-23:
> https://github.com/JoveToo/OctoPrint-MFA

Thanks for sharing. I will check that out later.

foosel, 2024-05-29:
FYI, there's now a draft PR that implements a new MFA plugin mixin type, and a [TOTP plugin utilizing that](https://github.com/OctoPrint/OctoPrint-MfaTotp). I still have to look into some E2E testing of the logic, and the docs also haven't yet seen an update, hence the draft status, but I still thought it might be interesting to poke at.

foosel, 2024-06-05:
This is now merged and will be part of 1.11.0.

foosel, 2025-04-22:
Closing this as 1.11.0 has just been released

closed 2025-04-22 by foosel