OctoPrint/OctoPrint #4935: [Request]

netinetwalker, 2024-01-23:
### Is your feature request related to a problem? Please describe.

prusa MK4 temperatures include heatbrake cooling fan response and target X. 
```
T:25.00/0.00 B:23.96/0.00 X:23.95/36.00 A:30.73/0.00 @:0 B@:0 HBR@:0
T:25.00/0.00 B:23.93/0.00 X:23.94/36.00 A:30.75/0.00 @:0 B@:0 HBR@:0
T:25.00/0.00 B:23.96/0.00 X:23.94/36.00 A:30.81/0.00 @:0 B@:0 HBR@:0
```

This causes some problems with plugins like: 
- OctoPrint PSU Control https://github.com/kantlivelong/OctoPrint-PSUControl/issues/271
- OctoPrint-PSUControl-TPLink https://github.com/kantlivelong/OctoPrint-PSUControl-TPLink/issues/8
- BetterHeaterTimeout
- HeaterTimeout

### Describe the solution you'd like

It would be nice to have a small filter setting in the printer profile for some temperatures

### Describe alternatives you've considered

At the moment I use a small filter plugin to fix the problem on my mk4:
```python
# coding=utf-8

def filter_heatbrake_temperatures(comm, parsed_temps):
    parsed_temps.pop("X", None)
    return parsed_temps

__plugin_name__ = "Filter Prusa MK4 heatbrake Temperatures"
__plugin_version__ = "0.0.1"
__plugin_description__ = "A quick heatbrake temperatures filter plugin for Prusa MK4"
__plugin_pythoncompat__ = ">=2.7,<4"
__plugin_hooks__ = {
    "octoprint.comm.protocol.temperatures.received": filter_heatbrake_temperatures
}
```

jneilliii, 2024-01-23:
I had to deal with this issue with my power control plugins. It's definitely a pain. I suspect though since this is only on an isolated number of printers (so far only MK4, Mini, XL) not sure if a core feature is needed here, and rather your single file plugin should be registered on the plugin repository for others to find.

netinetwalker, 2024-01-23:
IMHO in the future it would be good to split the heater and cooling temperatures and targes in separated items.   
One for all heaters one for all coolers ...
The old Interface provides only the heaters and the new function gives you the cooling Temps/Targets

Something like this?

```python
heaters = self._printer.get_current_temperatures()
coolers = self._printer.get_current_cooler_temperatures()
```

foosel, 2024-01-23:
I think in the future it would be good for 3d printer vendors to not make unilateral decisions about how to modify existing standard protocol messages - but that's not going to happen I guess, or it would already have done so 😉

As it's currently only one vendor, I'm leaning towards "plugin only" and the suggestion of @jneilliii 

Should this become something that's also available in vendor agnostic firmware variants, we can think about core support.

closed 2024-08-29 by jneilliii