OctoPrint/OctoPrint #4938: [RC Feedback] Feedback on the 1.10.0rc1 Release Candidate

foosel, 2024-01-31:
Please provide *general* feedback on your experience with the 1.10.0rc1 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

#### Things to take a closer look at

For this RC, these things should get a closer look while testing, if possible:

- Proper behaviour when using the included web interface as well as any third party clients at your disposal.
- User and group management functioning as expected.
- Plugin installation functioning as expected.
- Application key management functioning as expected. Authentication workflow with third party clients at your disposal (e.g. slicers) works as it should.

---

### ❗👇 IMPORTANT 👇❗
###
### If you run into any obvious bugs not yet listed below, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports *with system info bundles*, not just a statement that something doesn't work!
###
### ❗👆 IMPORTANT 👆❗

---

### Update test matrix

[image]

*Releases aren't rolled out unless all tests in the update test matrix succeed.*

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

### Issues needing analysis

b-morgan, 2024-01-31:
Updated two instances, OctoPi 0.18.0 and OctoPi 1.0.0cam.

b-morgan, 2024-01-31:
I guess I wasn't patient enough...

RPi3B, OctoPi 0.18.0, updated successfully.

ckuethe, 2024-01-31:
Basic stuff is working so far:
* homeassistant is correctly polling my printer
* camera still works
* arcwelder plugin still works
* printing is printing

Ender 5-S1, OctoPi 1.0.0 on RP3B+

The login banner is unable to report the version. I get this li'l traceback when I ssh in:

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
ImportError: cannot import name 'get_versions' from 'octoprint._version' (/home/pi/oprint/lib/python3.11/site-packages/octoprint/_version.py)
```

The files section seems to have moved. It used to be on the right side of the center panel. Or maybe I had some formatting plugin that isn't working...

[image]

ckuethe, 2024-01-31:
OctoRemote, Homeassistant, and PrusaSlicer are all able to operate over the API.

From my logs:
```
Jan 31 13:40:53 octopi octoprint[438]: Traceback (most recent call last):
Jan 31 13:40:53 octopi octoprint[438]:   File "/home/pi/oprint/lib/python3.11/site-packages/octoprint/plugins/softwareupdate/__init__.py", line 1852, in _get_current_version
Jan 31 13:40:53 octopi octoprint[438]:     information, is_current = version_checker.get_latest(
Jan 31 13:40:53 octopi octoprint[438]:                               ^^^^^^^^^^^^^^^^^^^^^^^^^^^
Jan 31 13:40:53 octopi octoprint[438]:   File "/home/pi/oprint/lib/python3.11/site-packages/octoprint/plugins/softwareupdate/version_checks/github_release.py", line 274, in get_latest
Jan 31 13:40:53 octopi octoprint[438]:     raise ConfigurationInvalid(
Jan 31 13:40:53 octopi octoprint[438]: octoprint.plugins.softwareupdate.exceptions.ConfigurationInvalid: Update configuration for arc_welder of type github_release needs all of user, repo and current set and not None

```

cp2004, 2024-01-31:
> The files section seems to have moved. It used to be on the right side of the center panel. Or maybe I had some formatting plugin that isn't working...

That would be some plugin that you had - OctoPrint doesn't have a right side bar.

> Jan 31 13:40:53 octopi octoprint[438]: octoprint.plugins.softwareupdate.exceptions.ConfigurationInvalid: Update configuration for arc_welder of type github_release needs all of user, repo and current set and not None

This was a bug in earlier versions of Arc Welder - I believe it has been fixed, but because it affected the software update process you have to manually update it. Reinstalling the plugin would work.

cp2004, 2024-01-31:
It would probably make sense to move it to use the (stable) APIs in `octoprint.util.version`, rather than importing from `octoprint._version`. It's a good observation that this has changed, I think it's unlikely it would have been used elsewhere given the documented API I mentioned, but you never know...

jneilliii, 2024-01-31:
[image]

no image in temp graph, even in safe mode.

jneilliii, 2024-02-01:
drag and drop upload seems to be broken for me in two instances, even running in safe mode. doesn't appear to be any errors in developer tools and the newly added setting is enabled.

sclebo05, 2024-02-01:
> Updated RPi4B (4GB), OctoPi 1.0.0cam successfully.
> 
> RPi3B, OctoPi 0.18.0 appears to be stuck:
> 
> ```
> Building wheel for zeroconf (PEP 517): still running...
> Building wheel for zeroconf (PEP 517): still running...
> ```
> 
> [OctoPrint_Update.zip](https://github.com/OctoPrint/OctoPrint/files/14117063/OctoPrint_Update.zip)

During my upgrade the following step was taking FOREVER:

/usr/bin/aarch64-linux-gnu-gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O2 -Wall -g -ffile-prefix-map=/build/python3.9-PN012d/python3.9-3.9.2=. -fstack-protector-strong -Wformat -Werror=format-security -g -fwrapv -O2 -fPIC -I/home/orangepi/OctoPrint/include -I/usr/include/python3.9 -c src/zeroconf/_protocol/incoming.c -o /tmp/pip-install-wkuqjzdp/zeroconf_eaa55c58057a47418ed13ad427f9a59a/build/temp.linux-aarch64-+
I used a combination of htop and 'top -bcn1 -w512' to determine which step was hanging.

The zeroconf wheel activity in the displayed log seems to reflect this delay with building zeroconf.  Once completed, I saw an error regarding OctoEverywhere, but OE seems to be working fine.  Camera, printer, etc seems OK.

FULL DISCLOSURE: I'm running an Orange Pi Zero 3 which is unsupported by Octoprint, so I understand if that nullifies my feedback.

[image]
[image]

foosel, 2024-02-01:
The issue with the welcome banner is now tracked in #4940 (it is technically an issue specific only to OctoPi, but since it happened due to code changes in OctoPrint it's a regression that needs looking into).

---

About the long build time of the zeroconf dependency... 

@sclebo05 I fear piwheels doesn't have builds for aarch64, so on that architecture you'll *have* to build it yourself, and that takes time. 

OctoPi 0.18 should by default however be using piwheels and the zeroconf wheel available on there, and in my update tests it did just that too, so I'm not entirely sure why that didn't work out for you @b-morgan and it had to pull down the sources and build the wheel itself...

I just did another flash of an 0.18 on the testrig and ran through the update again. Python 3.7.3, the wheel was pulled in from piwheels and no build was needed:

```
2024-02-01 09:03:13,723 > Collecting zeroconf~=0.127
2024-02-01 09:03:13,724 > Downloading https://www.piwheels.org/simple/zeroconf/zeroconf-0.131.0-cp37-cp37m-manylinux_2_28_armv7l.whl (3.2 MB)
```

In contrast, here's the related section from @b-morgan 's build:

```
Collecting zeroconf~=0.127
Downloading zeroconf-0.131.0.tar.gz (169 kB)
```

And that means, there it downloaded a source tarball from pypi, not the wheel from piwheels. @b-morgan could you please share a systeminfo bundle from the machine so I can take a closer look at the involved software versions and hopefully figure out what's going on? The only possible explanation I have right now is a Python version that's neither 3.7 nor 3.9 nor 3.11, [as only those wheels are available on piwheels](https://www.piwheels.org/project/zeroconf/).

(And I'm leaning more and more towards creating a `wheels.octoprint.org`)

---

About that OctoEverywhere issue, yeah, that's now a dependency mismatch between OctoPrint and OE. The issue with OE's version range there is that as I had to discover yesterday, `websocket-client` dropped Python 3.7 support in version 1.6.2 (yes, .2 - and no, there was no changelog entry on that). So if I depend on anything beyond that (and I actually didn't trust 1.6.1 to be Python 3.7 compatible, that drop of support in a point release felt too weird and like they just forgot to adjust the python requirement value in their `setup.py` until then, so I went with 1.5.3 instead), OctoPrint will no longer install on Python 3.7. And that means, anyone still on OctoPi 0.17 or 0.18 gets locked out of updates - according to the stats, that's two thirds of the OctoPi user base, around 80k instances.. 

At some point, that will happen of course (Python 3.7 is EOL now), but it wasn't something I wanted to do during roll-out of the first RC of 1.10.0 without any warning. It's something that can be addressed with some early heads-ups however, maybe even in the release notes of 1.10.0.

In general, Python versions dropping out of support is something I need to think up a solution for, in the shape of automatic heads-ups in the software maybe.

b-morgan, 2024-02-01:
Here's the bundle as requested.

[octoprint-systeminfo-20240201051335.zip](https://github.com/OctoPrint/OctoPrint/files/14125167/octoprint-systeminfo-20240201051335.zip)

foosel, 2024-02-01:
Hm... my stock 0.18.0 image, which is also the official release version (just confirmed with a sha512sum check against the img attached to the OctoPi release) ships with pip 20.3.3, yet your version is 20.2.4. And downgrading mine to that allows me to reproduce it:

``` 
// Confirm OctoPi version & downgrade pip

pi@octopia:~ $ cat /etc/octopi_version 
0.18.0
pi@octopia:~ $ ~/oprint/bin/pip install pip==20.2.4
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pip==20.2.4
[...]
Successfully installed pip-20.2.4
pi@octopia:~ $ ~/oprint/bin/pip --version
pip 20.2.4 from /home/pi/oprint/lib/python3.7/site-packages/pip (python 3.7)

// Install zeroconf package

pi@octopia:~ $ ~/oprint/bin/pip install zeroconf==0.131.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting zeroconf==0.131.0
  Downloading zeroconf-0.131.0.tar.gz (169 kB)
     |████████████████████████████████| 169 kB 972 kB/s 
  Installing build dependencies ... -^canceled
ERROR: Operation cancelled by user
[...]

// Thas was a source tarball about to be compiled, cancelled, now upgrade pip again

pi@octopia:~ $ ~/oprint/bin/pip install pip==20.3.3
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pip==20.3.3
[...]
Successfully installed pip-20.3.3
pi@octopia:~ $ ~/oprint/bin/pip --version
pip 20.3.3 from /home/pi/oprint/lib/python3.7/site-packages/pip (python 3.7)

// Install zeroconf package

pi@octopia:~ $ ~/oprint/bin/pip install zeroconf==0.131.0
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting zeroconf==0.131.0
  Downloading https://www.piwheels.org/simple/zeroconf/zeroconf-0.131.0-cp37-cp37m-manylinux_2_28_armv7l.whl (3.2 MB)
     |████████████████████████████████| 3.2 MB 973 kB/s 
[...]

// It now fetched the wheel from piwheels

```

So my guess is, you don't have an 0.18.0 release version there, but some earlier development build that still came with pip 20.2.4, and something between pip 20.2.4 and pip 20.3.3 changed, meaning 20.2.4 won't download (certain?) wheels from piwheels whereas 20.3.3 will.

I suggest to give pip an update ;)

In any case, this isn't an issue with stable OctoPi builds, so I hope it won't be something that many people run into.

b-morgan, 2024-02-01:
So that's what I get for running pre-release software 😁😎. Thanks to your post, I believe I upgraded my pip. Since my OctoPrint is already at 1.10.0rc1, I'm not sure I need to do anything else.

```
pi@octopi:~ $ ~/oprint/bin/pip install pip==20.3.3
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting pip==20.3.3
  Downloading pip-20.3.3-py2.py3-none-any.whl (1.5 MB)
     |████████████████████████████████| 1.5 MB 2.0 MB/s
Installing collected packages: pip
  Attempting uninstall: pip
    Found existing installation: pip 20.2.4
    Uninstalling pip-20.2.4:
      Successfully uninstalled pip-20.2.4
Successfully installed pip-20.3.3
pi@octopi:~ $
```

foosel, 2024-02-01:
No, you should be fine going forward (as far as we know 😅)

larp-welt, 2024-02-01:
> The files section seems to have moved. It used to be on the right side of the center panel. Or maybe I had some formatting plugin that isn't working...

Same to me! That's a problem with UI Customizer (0.1.9.9). 

My work around at the moment is, to make the left column one unit smaller.

Don't know, if I should report this here or to UI Customizer... :-/

jneilliii, 2024-02-02:
> Don't know, if I should report this here or to UI Customizer... :-/

That needs to be reported to that project since it's manipulating the UI.

cp2004, 2024-02-02:
> I'm not sure if this even rises to the level of a bug it's so minor, but I wanted to report it just in case.

This is exactly the right place to put it. A good observation to let us know a) people have tested it and b) that they have some feedback!

JohnOCFII, 2024-02-03:
OctoPrint 1.10.0rc1 on OctoPi 1.0.0 seems to be working well for me:

This is my first usage of OctoPrint on the Prusa MK4 after many years of using Octoprint in MK3 (and MK2 before that).  I had used PrusaLink -- and was underwhelmed.  I switched my PrusaSlicer back to creating GCODE and not BGCODE.  Link others, the ZeroConf build took a long time, but finished successfully.

- Uploaded new GCODE to OctoPrint
- Printed GCODE on Prusa MK4
- GCODE viewer worked well
- Terminal worked well
- Timelapse worked well
- OctoSlack plugin worked well
- Print Job History plugin worked well
- Announcements/Notifications in title bar worked well
- Classic Webcam and Camera settings all worked well
- Achievements Plug-in is fun!

wilmardo, 2024-02-08:
Just to add to this issue, I had to install `rust` and `binutils` for the update to the RC to complete.

It failed to compile the wheel for `maturin` without the rust compiler and binutils. This is new requirement since I have been running and updating without issues for over a year. I am on `armv7` btw.

Below are the logs, after the first one I installed rust `apt install rust`, after the second about the missing `arm-linux-androideabi-ar` I installed `binutils` by `apt install binutils`. The last one is successful install (that took 50 minutes :') )

<details>
  <summary>pip install logs, missing rust</summary>

```
2024-02-08 18:30:45,718 > Collecting https://github.com/OctoPrint/OctoPrint/archive/1.10.0rc1.zip
2024-02-08 18:30:46,731 > Downloading https://github.com/OctoPrint/OctoPrint/archive/1.10.0rc1.zip
2024-02-08 18:30:48,740 > / 6.4 MB 3.8 MB/s 0:00:01
2024-02-08 18:30:49,752 > Installing build dependencies: started
2024-02-08 18:31:02,818 > Installing build dependencies: finished with status 'done'
2024-02-08 18:31:02,820 > Getting requirements to build wheel: started
2024-02-08 18:31:04,863 > Getting requirements to build wheel: finished with status 'done'
2024-02-08 18:31:04,866 > Preparing metadata (pyproject.toml): started
2024-02-08 18:31:10,066 > Preparing metadata (pyproject.toml): finished with status 'done'
2024-02-08 18:31:10,069 > Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /data/data/com.termux/files/usr/lib/python3.11/site-packages (from OctoPrint==1.10.0rc1) (2021.2.23)
2024-02-08 18:31:10,071 > Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /data/data/com.termux/files/usr/lib/python3.11/site-packages (from OctoPrint==1.10.0rc1) (2021.10.11)
2024-02-08 18:31:10,072 > Collecting OctoPrint-PiSupport>=2023.10.10 (from OctoPrint==1.10.0rc1)
2024-02-08 18:31:10,074 > Downloading OctoPrint_PiSupport-2023.10.10-py2.py3-none-any.whl.metadata (2.9 kB)
2024-02-08 18:31:10,076 > Collecting argon2-cffi>=23.1.0 (from OctoPrint==1.10.0rc1)
2024-02-08 18:31:10,078 > Downloading argon2_cffi-23.1.0-py3-none-any.whl.metadata (5.2 kB)
2024-02-08 18:31:10,080 > Requirement already satisfied: Babel<2.13,>=2.12.1 in /data/data/com.termux/files/usr/lib/python3.11/site-packages (from OctoPrint==1.10.0rc1) (2.12.1)
2024-02-08 18:31:10,083 > Requirement already satisfied: cachelib<0.11,>=0.10.2 in /data/data/com.termux/files/usr/lib/python3.11/site-packages (from OctoPrint==1.10.0rc1) (0.10.2)
2024-02-08 18:31:10,085 > Collecting Click<9,>=8.1.7 (from OctoPrint==1.10.0rc1)
2024-02-08 18:31:10,087 > Downloading click-8.1.7-py3-none-any.whl.metadata (3.0 kB)
2024-02-08 18:31:10,089 > Requirement already satisfied: colorlog<7,>=6.7.0 in /data/data/com.termux/files/usr/lib/python3.11/site-packages (from OctoPrint==1.10.0rc1) (6.7.0)
2024-02-08 18:31:10,092 > Collecting emoji<3,>=2.10.1 (from OctoPrint==1.10.0rc1)
2024-02-08 18:31:10,092 > Downloading emoji-2.10.1-py2.py3-none-any.whl.metadata (5.3 kB)
2024-02-08 18:31:10,093 > Collecting feedparser<7,>=6.0.11 (from OctoPrint==1.10.0rc1)
2024-02-08 18:31:10,094 > Downloading feedparser-6.0.11-py3-none-any.whl.metadata (2.4 kB)
2024-02-08 18:31:10,094 > Requirement already satisfied: filetype<2,>=1.2.0 in /data/data/com.termux/files/usr/lib/python3.11/site-packages (from OctoPrint==1.10.0rc1) (1.2.0)
2024-02-08 18:31:10,095 > Collecting Flask-Assets<3,>=2.1.0 (from OctoPrint==1.10.0rc1)
2024-02-08 18:31:10,096 > Downloading Flask_Assets-2.1.0-py3-none-any.whl.metadata (931 bytes)
2024-02-08 18:31:10,096 > Requirement already satisfied: Flask-Babel<4,>=3.1.0 in /data/data/com.termux/files/usr/lib/python3.11/site-packages (from OctoPrint==1.10.0rc1) (3.1.0)
2024-02-08 18:31:10,097 > Collecting Flask-Login<0.7,>=0.6.3 (from OctoPrint==1.10.0rc1)
2024-02-08 18:31:10,098 > Downloading Flask_Login-0.6.3-py3-none-any.whl.metadata (5.8 kB)
2024-02-08 18:31:10,098 > Collecting Flask-Limiter<4,>=3.5.0 (from OctoPrint==1.10.0rc1)
2024-02-08 18:31:10,099 > Downloading Flask_Limiter-3.5.0-py3-none-any.whl.metadata (6.3 kB)
2024-02-08 18:31:10,099 > Requirement already satisfied: flask<2.3,>=2.2.3 in /data/data/com.termux/files/usr/lib/python3.11/site-packages (from OctoPrint==1.10.0rc1) (2.2.3)
2024-02-08 18:31:10,100 > Collecting frozendict<3,>=2.4.0 (from OctoPrint==1.10.0rc1)
2024-02-08 18:31:10,101 > Downloading frozendict-2.4.0.tar.gz (314 kB)
2024-02-08 18:31:10,101 > ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 314.6/314.6 kB 3.7 MB/s eta 0:00:00
2024-02-08 18:31:10,102 > Installing build dependencies: started
2024-02-08 18:31:22,254 > Installing build dependencies: finished with status 'done'
2024-02-08 18:31:22,255 > Getting requirements to build wheel: started
2024-02-08 18:31:23,465 > Getting requirements to build wheel: finished with status 'done'
2024-02-08 18:31:23,466 > Installing backend dependencies: started
2024-02-08 18:31:30,486 > Installing backend dependencies: finished with status 'done'
2024-02-08 18:31:30,486 > Preparing metadata (pyproject.toml): started
2024-02-08 18:31:32,715 > Preparing metadata (pyproject.toml): finished with status 'done'
2024-02-08 18:31:32,715 > Requirement already satisfied: future<1,>=0.18.3 in /data/data/com.termux/files/usr/lib/python3.11/site-packages (from OctoPrint==1.10.0rc1) (0.18.3)
2024-02-08 18:31:32,716 > Collecting markdown<3.5,>=3.4.4 (from OctoPrint==1.10.0rc1)
2024-02-08 18:31:32,716 > Downloading Markdown-3.4.4-py3-none-any.whl.metadata (6.9 kB)
[... cut ...]
2024-02-08 21:44:07,674 > Uninstalling OctoPrint-1.9.3:
2024-02-08 21:44:07,674 > Successfully uninstalled OctoPrint-1.9.3
2024-02-08 21:44:14,720 > Successfully installed Click-8.1.7 Flask-Assets-2.1.0 Flask-Limiter-3.5.0 Flask-Login-0.6.3 OctoPrint-1.10.0rc1 OctoPrint-PiSupport-2023.10.10 argon2-cffi-23.1.0 blinker-1.6.3 emoji-2.10.1 feedparser-6.0.11 frozendict-2.4.0 markdown-3.4.4 netifaces2-0.0.21 psutil-5.9.8 requests-2.31.0 sentry-sdk-1.40.2 websocket-client-1.5.3 zeroconf-0.131.0 zipstream-ng-1.7.1
```

</details>

foosel, 2024-02-12:
@wilmardo On what system/platform is that, is that Octo4a or some custom install under Termux? 

That dependency on `maturin` and a working Rust compile environment got pulled in by the switch to `netifaces2` as a replacement for the abandoned and now unmaintained `netifaces`. If no wheel for that can be found (which for your runtime environment it can't), it'll have to build the native bits, and for that it needs `maturin` and Rust.

foosel, 2024-02-12:
@nastevens 

> I'm not sure if this even rises to the level of a bug it's so minor, but I wanted to report it just in case.

I've played around with it some more now and have come to the conclusion that it's so minor we are going to ignore that. I wasn't able to reproduce the amount of "jitter" that you recorded, I could see *some* jittering, but in neither Firefox nor Chromium was it bad enough to look deeper.

It would also not be something easily approached, given that the dashing option for the graph here is provided by [a flot plugin](https://github.com/cquartier/flot.dashes), so this would then have to either be fixed manually in the vendored js and/or reported upstream.

But at least this just made me look into the bugs on the plugin, which in turn made me find out that contrary to the GitHub releases shown on flot, there HAVE been newer releases in the last years after all, and apparently flot now supports dashed graphs by default. So I'll take a look at updating flot (not something I want to do on this RC, there's probably going to be a lot of breakage going to happen), and maybe that other implementation of dashed graphs won't have that issue.

In any case, currently indeed very minor and not going to get tackled for 1.10.0, but it's still good that you reported it! That's what RCs are for!

wilmardo, 2024-02-12:
> On what system/platform is that, is that Octo4a or some custom install under Termux?

The latter, I run octoprint on a OnePlus One with a custom kernel (for serial drivers and charging and OTG at the same time). Good point, I should have added it since I know I am fully unsupported :)

> That dependency on maturin and a working Rust compile environment got pulled in by the switch to netifaces2 as a replacement for the abandoned and now unmaintained netifaces. If no wheel for that [...]

Thanks for the clarification! My goal with my reply was just to add the information if someone else might run into this (older Raspberry Pi's might be the same arch).

foosel, 2024-02-12:
1.10.0rc2 is out, new feedback ticket is here: #4948

closed 2024-02-12 by foosel