OctoPrint/OctoPrint #4939: [1.10.0rc1] drag and drop upload not working

jneilliii, 2024-02-01:
### The problem

When dragging a gcode file onto the UI the overlay doesn't show and upload never happens. 

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.0rc1

### Operating system running OctoPrint

OctoPi 1.1.0 (nightly bookworm build) and OctoPi 0.18.0 (build 2022.10.18.093204)

### Printer model & used firmware incl. version

Virtual Printer

### Browser and version of browser, operating system running browser

Google Chrome Version 121.0.6167.87 (Official Build) (64-bit)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240131192605.zip](https://github.com/OctoPrint/OctoPrint/files/14118792/octoprint-systeminfo-20240131192605.zip)

[192.168.126.109-1706747235133.log](https://github.com/OctoPrint/OctoPrint/files/14118800/192.168.126.109-1706747235133.log)

https://github.com/OctoPrint/OctoPrint/assets/5249455/6903086f-1ddc-4204-9fa2-2d2ff75343d3

larp-welt, 2024-02-01:
I have the same problem...

[octoprint-systeminfo-20240201233649.zip](https://github.com/OctoPrint/OctoPrint/files/14132206/octoprint-systeminfo-20240201233649.zip)

foosel, 2024-02-02:
I'll look at this on Monday :+1:

jaymcd, 2024-02-03:
Same issue here

jneilliii, 2024-02-03:
I think this may be a race condition with settings load. My previous PR that enabled the option in settings was definitely working, but now seems that the `feature_enableDragDropUpload` observable is detected as undefined on serverConnect when the drag and drop binding is taking place.

jneilliii, 2024-02-03:
possible solution might be to do the `self._enableDragNDrop(true);` in an observable subscription to `feature_enableDragDropUpload`, and pass along the boolean from new value?

jaymcd, 2024-02-03:
In my case, will test if it had anything to do with UI customizer as that seems to be acting weird well. Many settings don't take.Jay

jneilliii, 2024-02-04:
It doesn't make a difference with UI Customizer or not. The issue exists in safe mode, which has no plugins running.

foosel, 2024-02-12:
1.10.0rc2 has just been released :+1:

commit 64446d9c (2024-02-05): 🐛 Fix drag'n'drop initialization

Fixes #4939
closed 2024-02-12 by foosel