OctoPrint/OctoPrint #4940: [1.10.0rc1] New versioning module doesn't support CLI welcome banner code deployed with OctoPi

foosel, 2024-02-01:
### Problem

> The login banner is unable to report the version. I get this li'l traceback when I ssh in:
> 
> ```
> Traceback (most recent call last):
>   File "<string>", line 1, in <module>
> ImportError: cannot import name 'get_versions' from 'octoprint._version' (/home/pi/oprint/lib/python3.11/site-packages/octoprint/_version.py)
> ```

_Originally posted by @ckuethe in https://github.com/OctoPrint/OctoPrint/issues/4938#issuecomment-1920046525_

### Solution

Two options come to mind:

1) Extend the new custom versioning tool to be API compatible, as in, bring back `get_versions`
2) Adjust the OctoPi welcome banner

The problem with 2) is that we can only do that going forward, not retroactively on currently installed images, so the welcome banner seen when SSHing in would be broken there.

Hence, I guess both needs to be done - adjust the welcome banner to use the available functions in `octoprint.util.version` going forward, but also bring back `get_versions` so we don't have issues with existing installations in the field.

foosel, 2024-02-01:
OctoPrint 1.10.0rc2 will bring back `octoprint._version.get_versions`. It's not API compatible, but at least the part of it needed by the welcome banner on the OctoPi instances out there should work.

And OctoPi-UpToDate has gotten an addition that will take care of fixing things there to use `octoprint.util.version` instead.

ckuethe, 2024-02-01:
Thanks for looking into this! It wasn't a big problem, it didn't get in the way of actually printing, but it seemed worth mentioning.

jneilliii, 2024-02-02:
This is a good idea, because there are a couple of plugins that are utilizing get_version for dynamically loading stuff...for example:

```
2024-02-01 11:21:43,159 - octoprint.plugin - ERROR - Error while calling plugin PrintTimeGenius
Traceback (most recent call last):
  File "C:\OctoPrint\WPy64-31050\python-3.10.5.amd64\lib\site-packages\octoprint\plugin\__init__.py", line 275, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "C:\OctoPrint\WPy64-31050\python-3.10.5.amd64\lib\site-packages\octoprint\util\__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "C:\OctoPrint\WPy64-31050\python-3.10.5.amd64\lib\site-packages\octoprint_PrintTimeGenius\__init__.py", line 583, in on_startup
    if pkg_resources.parse_version(octoprint._version.get_versions()['version']) == pkg_resources.parse_version("1.3.9rc2"):
AttributeError: module 'octoprint._version' has no attribute 'get_versions'
```

foosel, 2024-02-12:
1.10.0rc2 has just been released :+1:

commit 48586722 (2024-02-01): 🐛 Make octoprint._version.get_versions work again

The welcome banner on OctoPi's CLI still uses that instead
of octoprint.util.version

Closes #4940
closed 2024-02-12 by foosel