OctoPrint/OctoPrint #4941: [1.10.0rc1] Appkeys plugin logs error

crysxd, 2024-02-02:
### The problem

Requesting `/plugin/appkeys/probe` triggers the below error, it seems the plugin is still working as expected though (both probe and request)

```
2024-02-02 13:16:08,548 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin appkeys
Traceback (most recent call last):
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/api/__init__.py", line 97, in pluginData
    response = api_plugin.on_api_get(request)
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/plugins/appkeys/__init__.py", line 338, in on_api_get
    keys=list(
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/plugins/appkeys/__init__.py", line 340, in <lambda>
    lambda x: x | {"api_key": ""},
 ```

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.0rc1

### Operating system running OctoPrint

Linux

### Printer model & used firmware incl. version

/

### Browser and version of browser, operating system running browser

/

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240202133130.zip](https://github.com/OctoPrint/OctoPrint/files/14140658/octoprint-systeminfo-20240202133130.zip)

jneilliii, 2024-02-02:
adding full message error...for better visibility.

```
2024-02-02 13:27:00,601 - octoprint.server.util.sockjs - INFO - New connection from client: 100.120.245.127
2024-02-02 13:27:00,622 - octoprint.server.util.flask - INFO - Passively logging in user beagle from 100.120.245.127
2024-02-02 13:27:00,622 - octoprint.access.users - INFO - Logged in user: beagle
2024-02-02 13:27:00,661 - octoprint.plugins.softwareupdate - INFO - Fetching check overlays from https://plugins.octoprint.org/update_check_overlay.json
2024-02-02 13:27:01,530 - octoprint.server.api - ERROR - Error calling SimpleApiPlugin appkeys
Traceback (most recent call last):
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/server/api/__init__.py", line 97, in pluginData
    response = api_plugin.on_api_get(request)
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/plugins/appkeys/__init__.py", line 338, in on_api_get
    keys=list(
  File "/octoprint/plugins/lib/python3.8/site-packages/octoprint/plugins/appkeys/__init__.py", line 340, in <lambda>
    lambda x: x | {"api_key": ""},
TypeError: unsupported operand type(s) for |: 'dict' and 'dict'
```

foosel, 2024-02-02:
Ooh, I think I know what happened. I did my development under python 3.11, and this operation is most likely not available on older versions yet and I completely forgot that.

foosel, 2024-02-05:
Yeah, should have looked closer 😅

[image]

foosel, 2024-02-05:
This should be fixed by the above commit.

foosel, 2024-02-12:
1.10.0rc2 has just been released :+1:

commit 4159c8c5 (2024-02-05): 🐛 Fix syntax under Python 3.7 & 3.8

They don't yet support `dict | other_dict` syntax, so let's do this
differently.

Fixes #4941
closed 2024-02-12 by foosel