OctoPrint/OctoPrint #4942: [1.10.0rc1] Virtual printer can't be disabled

crysxd, 2024-02-02:
### The problem

Some of my OctoPrint instances are updated to 1.10.0rc1 and all those instances can't disable the virtual printer. The instances that are still on 1.9.3 do not show this behavior. As it stands it seems impossible to disable the virtual printer. I'm not sure if the virtual printer was enabled at the time of the update and if this triggers the issue. I did not find anything in the logs after changing the setting.

https://github.com/OctoPrint/OctoPrint/assets/5859228/452bfa88-bd36-4615-bde2-993cb2953ba8

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.0rc1

### Operating system running OctoPrint

Linux

### Printer model & used firmware incl. version

/

### Browser and version of browser, operating system running browser

Chrome / Postman

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240202133130.zip](https://github.com/OctoPrint/OctoPrint/files/14140707/octoprint-systeminfo-20240202133130.zip)

foosel, 2024-02-05:
That bug was caused by the performance improving changes in the `HierarchicalChainMap`, it hadn't yet covered a specific case. I've now added a test for that, confirmed the test failed before, then added a fix. I hope this didn't break anything else currently not covered by tests, if so we'll need to add more cases to the test suite.

crysxd, 2024-02-05:
Will wait for RC2 and try again :) Thank you for the fix!

foosel, 2024-02-12:
1.10.0rc2 has just been released :+1:

commit cebb0880 (2024-02-05): 🐛 Fix handling of setting empty dict on config

Should delete prefix, so far didn't.

Fixes #4942
closed 2024-02-12 by foosel