OctoPrint/OctoPrint #4943: [1.10.0rc1] No warning when overwriting a file on upload

larp-welt, 2024-02-02:
### The problem

When I upload a gcode file with the same name as an existing one, no window message pops up to warn me and to ask me what to do.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.10.0rc1

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Pursa Mk3S+

### Browser and version of browser, operating system running browser

Version 120.0.6099.199 (Offizieller Build) for TUXEDO Computers GmbH (64-Bit) running on Tuxedo

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

https://github.com/OctoPrint/OctoPrint/assets/392672/6f719c5e-ae54-428c-aeb3-df45aa488c78

[octoprint-systeminfo-20240202194250.zip](https://github.com/OctoPrint/OctoPrint/files/14144862/octoprint-systeminfo-20240202194250.zip)
[console.log](https://github.com/OctoPrint/OctoPrint/files/14144879/console.log)

jneilliii, 2024-02-03:
This seems to be reproducible inside a folder. Uploading to the root folder I do get the overwrite/rename prompt as expected.

jneilliii, 2024-02-03:
Along with that, if a file exists in the root uploads folder, attempting to upload the same named file in a folder shows the overwrite/rename prompt.

foosel, 2024-02-05:
That was an error crept in through the PR that added the file details and which I overlooked during review, sh*t happens ^^

Now fixed for 1.10.0rc2, thanks for reporting!

foosel, 2024-02-12:
1.10.0rc2 has just been released :+1:

commit 311faebb (2024-02-05): 🐛 Fix fetching of file details for existence check

That used to be correct and got changed by a PR and
overlooked in review, 💩 happens.

Fixes #4943
closed 2024-02-12 by foosel