OctoPrint/OctoPrint #4947: "Fake Acknowledgement" required to get past bad serial parses?

nbergen, 2024-02-11:
### The problem

My setup has been running without issue for a bit more than a year.  Recently, Octoprint was inadvertently updated from 1.8.3 to 1.9.3, and I began having issues with prints.

The issue manifests by hanging the print until the communication timeout expires.  This increases the time to print as well as ruining the print with blobbing as the printhead remains motionless for a few seconds.  

It can be partially, though insufficiently mitigated by both reducing the printer communication timeout, and running a modified version of the old octoprint_ackackack plugin:
https://gist.github.com/nbergen/abb07a55df30b05f6aa8ce6e2d771aa3

It appears to have to do with serial parsing, as sometimes concatenated lines can plainly be seen.
[image]

Other times, there appears to be no OK in the stream.  This commonly happens immediately after starting the print, it transitions to "Starting", but will not set the target temperature until "Fake Acknowledgement" is pressed (or the case is manually handled in octoprint_ackackack).

I've found a handful of similar issues of course, both in Marlin and Octoprint, but none appear to match exactly with what I'm seeing.
https://community.octoprint.org/t/serial-connection-commands-bufferend-until-fake-ack/11839/7
https://github.com/OctoPrint/OctoPrint/issues/2454
https://github.com/MarlinFirmware/Marlin/issues/26507
https://www.reddit.com/r/VoxelabAquila/comments/rwoovl/h32_blobbing_with_only_octoprint_or_all_usb/

I do see a note in #2454 that this is suspected to be an issue in the underlying USB/Serial processing libraries, though I was fine before the upgrade to 1.9.3.  The issue now reproduces in all Octoprint versions I've tested from 1.8.2 through 1.10.3rc, which makes me wonder if some other supporting library was upgraded with the Octoprint upgrade that's causing the issue.

I am running an old version of Marlin's bugfix branch, and I haven't ventured into troubleshooting the firmware, though I have read the notes on increasing the buffer size, as
1. I was stable with this firmware for more than a year.
2. I no longer have the dev environment set up to build the firmware.
3. It was rather finicky to find a config the printer was happy with.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.3

### Operating system running OctoPrint

Ubuntu 20.04.5 LTS

### Printer model & used firmware incl. version

Ender5: MarlinFirmware/Marlin:bugfix-2.1.x@5809bdd (old marlin release)

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240211003037.zip](https://github.com/OctoPrint/OctoPrint/files/14230910/octoprint-systeminfo-20240211003037.zip)

[serial.log](https://github.com/OctoPrint/OctoPrint/files/14230911/serial.log)

```
Less has finished and no sheets were loaded.
packed_core.js?d7656e66:1755 Starting dependency resolution...
packed_core.js?d7656e66:1768 ... dependency resolution done
packed_core.js?d7656e66:1791 Initial application setup done, connecting to server...
packed_core.js?d7656e66:1585 Connected to the server
packed_core.js?d7656e66:1586 ❗ Safe mode is active. Third party plugins and language packs are disabled and cannot be enabled.
packed_core.js?d7656e66:1586 ❗ Reason for safe mode: settings
packed_core.js?d7656e66:1590 Triggering connect on all view models
packed_core.js?d7656e66:1592 Server (re)connect processed
packed_core.js?d7656e66:1791 Finalizing application startup
packed_core.js?d7656e66:1772 Going to bind 38 view models...
packed_core.js?d7656e66:1784 Did not bind view model BackupViewModel to target #wizard_plugin_backup since it does not exist
packed_core.js?d7656e66:1784 Did not bind view model ClassicWebcamWizardViewModel to target #wizard_classicwebcam since it does not exist
[... 116 lines cut ...]
sentry-sdk                    1.40.3
setuptools                    44.0.0
sgmllib3k                     1.0.0
Sidebar-Webcam                0.1.8
six                           1.16.0
sniffio                       1.3.0
SQLAlchemy                    1.1.18
Terminal-Commands-Extended    0.1.7
tinydb                        4.7.0
tornado                       6.4
typing_extensions             4.9.0
Unidecode                     1.3.8
uptime                        3.0.1
uritools                      2.1.1
urllib3                       2.2.0
watchdog                      2.3.1
webassets                     2.0
websocket-client              1.7.0
Werkzeug                      2.2.3
wheel                         0.42.0
wrapt                         1.15.0
zeroconf                      0.39.4
zipp                          3.17.0
zipstream-ng                  1.7.1
```

foosel, 2024-02-12:
Your softwareupdate log should contain everything that got updated along side OctoPrint with regards to Python packages, so that's where I'd start to dig. Unfortunately the one contained in your bundle only contains a plugin update from Feb 10th. The update log shows only one update of OctoPrint from 1.9.2 to 1.10.0rc1, no information seems to be available about your mentioned update from 1.8.3 to 1.9.3.

Can you please verify that you've shared the bundle of the correct instance? Do you maybe still have the software update log that contains the update you went through from 1.8.3 to 1.9.3?

What I can see in the commit log is that I upgraded the dependency on pyserial from 3.4 to 3.5 between 1.8.x and 1.9.0. What you could try is rolling back to pyserial 3.4 (`pip install pyserial==3.4` - this will probably log a big warning about unmet dependencies then since OctoPrint states it expects 3.5, but it should still work I hope) and see if that improves things again.

If it is indeed the pyserial update that triggered that (and that's tbh the only thing I could imagine having an impact here), it would match your observation that downgrading OctoPrint didn't solve it, as that wouldn't downgrade pyserial alongside it as long as its current version is within the stated version range, which 3.5 indeed is even for OctoPrint 1.8.3.

foosel, 2024-02-12:
Oh, also, do you happen to have "Request low latency mode" in Settings > Serial Connection > Connection?

foosel, 2024-03-18:
@nbergen I'm still waiting on more information from you here.

nbergen, 2024-03-18:
> @nbergen I'm still waiting on more information from you here.

👍 I'll have some time next week to chase it down.

nbergen, 2024-04-07:
> The update log shows only one update of OctoPrint from 1.9.2 to 1.10.0rc1, no information seems to be available about your mentioned update from 1.8.3 to 1.9.3.

Unfortunately I uninstalled entirely before reinstalling in a final scorched-earth fix attempt, convinced it would definitely work given the historical stability.  I suspect this wiped out the useful logs.  

[image]

To be honest, I don't know what I was thinking -- I should have spun up a new VM.  Sorry for that. Presently it's running 1.9.3

I downgraded pyserial to 3.4, observing the expected error message, but it did appear to work.
[image]

After restarting octoprint from the UI, I saw artifacts in the temperature polling
[image]

This could be a red herring, but the temperature view appears to be malfunctioning after starting a calibration cube print. The progress bar also appears to be incorrect, speeding through the first 20-30% of the print. I had to punch "Fake Awknowledgement" a couple of times to get the print started, but I didn't see a pause/glob in the first few minutes of watching.  I'll allow it to proceed, and report back with what happens, I can provide another set of the logs once the print completes.
[image]
[image]

I'll also get a new VM up with a true scorched-earth install with latest and see if that doesn't fix everything. This all seems very odd; I don't know what I did to get the VM in such an odd state.

nbergen, 2024-04-07:
It looks like it was just appearing to work as a result of my ackackack handling most of the corruption cases.

Eventually, I hit an unrecoverable message (no 'ok' at all!), and there's nothing that can be done to detect it.
[image]
*** a short timeout later ***
[image]

I'll proceed with the new-VM install, keeping this one around for future debugging if desired.

foosel, 2024-04-07:
Screenshots of the terminal sadly won't allow me to dig into this, please enable `serial.log`, reproduce the issue and then share a bundle. That will give me the `serial.log` and a fresh `octoprint.log` as well.

Especially considering you have to hit "fake acknowledgement", I'll need to see what exactly is happening on the serial communication leading up to that. 

> After restarting octoprint from the UI, I saw artifacts in the temperature polling

What artifacts? All I see at first glance is regular automatic temperature reports, interspersed with ones resulting from the sent `M105`.

Is this running in safe mode? If not I'd suggest to change that, to rule out any interference from third party plugins. The completely empty temperature graph is something I've never seen with temperature reports looking normal, so something is seriously off with your install there indeed.

You should never be needing the ackackack with a properly functioning environment. If `ok`s get interspersed with other responses, or outright vanish, there's something wrong with the serial communication way beneath OctoPrint.

To be honest, this all more and more looks rather like something that should be taken to the community forums rather than the bug tracker, as it seems to be something *very* specific to your environment. Most of the instances out there are on 1.9.3, I haven't seen any other reports that indicated issues with Pyserial 3.5, and thus I don't see anything actionable here from OctoPrint's side.

nbergen, 2024-04-07:
Sure, I certainly understand the importance of actionable diagnostics data, but did want to provide a clear picture of the mess I made in case someone stumbles across this thread in the future. 

Discarding this environment in favor of a fresh setup does seem like the path forward -- this does seem like an environmental problem with this VM, and I'll close this issue accordingly.

If I do see a repro in a clean environment, I'll get the bundle data to you. I appreciate the time you spent looking into it.

closed 2024-04-07 by nbergen

nbergen, 2024-05-19:
Following up to confirm that a fresh VM & install eliminated the problem.