OctoPrint/OctoPrint #4948: [RC Feedback] Feedback on the 1.10.0rc2 Release Candidate

foosel, 2024-02-12:
Please provide *general* feedback on your experience with the 1.10.0rc2 Release Candidate here. An "All is working fine" is valuable feedback as well, because it tells me that people in fact are testing the RC and just not finding any problems. Thanks :)

#### Things to take a closer look at

For this RC, these things should get a closer look while testing, if possible:

- Proper behaviour when using the included web interface as well as any third party clients at your disposal.
- User and group management functioning as expected.
- Plugin installation functioning as expected.
- Application key management functioning as expected. Authentication workflow with third party clients at your disposal (e.g. slicers) works as it should.
- Backup creation, download and restore functioning as expected :new:

---

### ❗👇 IMPORTANT 👇❗
###
### If you run into any obvious bugs not yet listed below, please *open a new ticket and follow ["How to file a bug report"](https://github.com/OctoPrint/OctoPrint/blob/master/CONTRIBUTING.md#how-to-file-a-bug-report)*. I need full bug reports *with system info bundles*, not just a statement that something doesn't work!
###
### ❗👆 IMPORTANT 👆❗

---

### Currently known issues

### Unreproduced issues

### Unreproduced and information for further analysis missing

### Third party issues

- zeroconf dependency gets built from source instead of pulled from piwheels on OctoPi 0.18, see https://github.com/OctoPrint/OctoPrint/issues/4938#issuecomment-1919869375
  - caused by an outdated pip version, [see this comment](https://github.com/OctoPrint/OctoPrint/issues/4938#issuecomment-1921262160)
  - if affected (OctoPi 0.18.0 development build instead of stable?), make sure to update to `pip>=20.3`
- UI Customizer Plugin: 12-column layout no longer fits, camera has additional black borders

### Issues needing analysis

foosel, 2024-02-12:
Don't be alarmed by the build for this release being red, I ran into an issue due to some CI changes while pushing the release to PyPI that I'll (hopefully) fix with 1.10.0rc3 or stable (whatever comes first).

jneilliii, 2024-02-12:
not sure if this reproducible yet, but I think I saw the same on the previous rc. When going to get more from plugin manager you are prompted to re-auth, but after entering password the reload page overlay instantly shows rather than just logging you in.

tigercjn, 2024-02-13:
I upgraded from RC1 to RC2 to do a reprint of my latest part; it would not connect using Auto but did via USB0, and it could not load and print from the stored file. I downgraded to 1.9.3 ok, and it is printing fine. I will try later to upgrade from here to rc2 and try and provide more information.

tigercjn, 2024-02-13:
> I upgraded from RC1 to RC2 to do a reprint of my latest part; it would not connect using Auto but did via USB0, and it could not load and print from the stored file. I downgraded to 1.9.3 ok, and it is printing fine. I will try later to upgrade from here to rc2 and try and provide more information.

Upgrading from 1.93 to 1.10.0rc2 worked fine and everything seems as usual.

foosel, 2024-02-13:
> not sure if this reproducible yet, but I think I saw the same on the previous rc. When going to get more from plugin manager you are prompted to re-auth, but after entering password the reload page overlay instantly shows rather than just logging you in.

Hm, I can't say I've ever seen this so far.

The only way this could happen that I can imagine is that it would have to lose the connection to the server briefly and then immediately reconnect, detecting version, installed plugins or most likely something in the config changed and then this would be shown.

Entering the password will trigger an active login in the background, so an additional request to the server, so *maybe* under very specific timing or situation that could mean a short break in the connection? No idea how to trigger this on purpose however.

When you saw this, did the reauthentication dialog go away, THEN the reload overlay showed, or did the reload overlay showed up first, underneath the reauthentication dialog, and then it closed?

jneilliii, 2024-02-13:
> When you saw this, did the reauthentication dialog go away, THEN the reload overlay showed, or did the reload overlay showed up first, underneath the reauthentication dialog, and then it closed?

I'll try again tonight after work to see if it happens again. If I remember correctly the reauth dialog showed and then after entering password the reload overlay popped.

TLMcNulty, 2024-02-19:
Flashed 3 SD cards this week, ended up on rc2 twice. General use, settings, printing workflow, essentials worked well. Currently running a newly flashed 1.9.3 from the camera streamer branch because I can't get my camera working correctly. It's an rpi cam v3. I replaced and reseated the cable, and it's being recognized and streaming correctly, but I can't focus it. This was consistent across 1.9.3 and the RCs. I'm still suspicious of software but I doubt the hardware as well now. I flicked it pretty hard and I think it focused for a sec. I'm not really sure what's up. 

Rebooting the pi crashes my printer right now though. I have a Prusa Mk4 and I'd like to hardwire the pi from the printer's 24v, but if OctoPi starts while connected to the printer it forces the printer into a crash/alarming with an idle message. This means any time I reboot the pi the printer panics and I have to reset it. On the bench it was nice because it was an audible alarm that the pi had booted but now that I have to walk down two flights to reset when rebooting and testing fixing my focus. The serial port is offline after the error so I guess it doesn't like that port disappearing or whatever. I searched for issues/advice on the above but didn't find anything. 

Reboot crash happened on the RCs as well I think. I can't easily test it as I write this but if it'd be helpful I can flash and upgrade to an RC to confirm. Ah just noticed some of your initial points

- Proper behaviour when using the included web interface as well as any third party clients at your disposal.
- User and group management functioning as expected.
Yes. Login, wizard, account creation, serial detection, printing work. 
- Plugin installation functioning as expected.
Yes, both officially bundled and public browsing. Only tested with blacklisting enabled.
- Application key management functioning as expected. Authentication workflow with third party clients at your disposal (e.g. slicers) works as it should.
Did not test.
- Backup creation, download and restore functioning as expected 🆕
Did not test. Can start testing if I keep flashing as much as I am.

cp2004, 2024-02-19:
@TLMcNulty there is an issue open on Prusa's firmware for the MK4 that was describing a crash when a USB host was connected. https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/3584

- although it now seems to claim it will be fixed with new printer firmware. Haven't followed, just aware of it.

TLMcNulty, 2024-02-20:
Ah, thanks, I should've mentioned that. I forgot I messed with that after the build, been focused more on octoprint. I found that and flashed the newest firmware I could, but the version at the time was worse and made it unusable. I might have some time between prints to flash the printer again.

jneilliii, 2024-02-24:
Curious what the expected outcome of the new re-authentication mechanism relative to the following config? Since there is no actual user created manually the password was never set due to addRemoteUsers. Since I don't know the password I can't change it either as the end user. 

```
accessControl:
  addRemoteUsers: true
  remoteUserHeader: whatever-header
  trustRemoteUser: true
```

sarusani, 2024-03-03:
Notification received via `//action:notification Text` are not shown in 1.10.0.
Works in 1.9.3

jneilliii, 2024-03-03:
> Notifications received via `//action:notification Text` are not shown in 1.10.0.
> Works in 1.9.3

confirmed, same issue for me.

ckuethe, 2024-03-04:
Works for me. The login banner on octopi now reports the correct version ("OctoPrint version : 1.10.0rc2").

Printed a few things this afternoon and it all just worked:
* OctoRemote on android could talk to my OctoPi
* HomeAssistant could pull print statistics over the API
* PrusaSlicer could send print jobs over the API
* timelapses generated correctly
* arcwelder generated G2/G3 instructions as expected

Creality Ender 5S1, stock firmware

Completely uneventful update - just the way I like it :)

foosel, 2024-03-18:
1.10.0rc3 has just been released.