OctoPrint/OctoPrint #4951: [Request] extend onBeforePrintStart to include selected file data

jneilliii, 2024-02-17:
### Is your feature request related to a problem? Please describe.

While developing the Bambu plugin I was attempting to use the onBeforePrintStart viewmodel callback to pop my own print options modal prior to starting. If you use the load and print button from the file list the filesViewModel.selectedfile property is not populated with the file that the button was pressed from but the previously selected file if any. This means I have no way of knowing which file was selected to print. 

### Describe the solution you'd like

Would like to get the selected file data object (including name, path, storage, refs, etc.) added to the onBeforePrintStart callback to allow for further processing.  

### Describe alternatives you've considered

* monkey patching filesViewModel.loadfile (probably will be my interim solution)
* reordering the loadfile function to have the callbacks after file selection so that filesViewModel.selectedfile is populated prior
* disabling the load and print button entirely and force selecting the file

foosel, 2024-10-28:
Implemented by #5073 and ready for 1.11.0, thanks! :+1:

foosel, 2025-04-22:
Closing this as 1.11.0 has just been released

closed 2025-04-22 by foosel