OctoPrint/OctoPrint #4952: Can not upload multiple files if "Upload to SD" (aka streaming) is disabled via uncheckiing "Enable SD supprt"

solidus0079, 2024-02-17:
### The problem

1.  I wanted to disable the "Upload to SD" button (aka streaming gcode) since I click it by accident sometimes and it takes forever.
2.  That button is removed by unchecking "Enable SD support" in Settings -> Features
3.  With "Enable SD support" unchecked, you can no longer upload multiple .gcode files at once via the remaining "Upload" button.  Multiple files at once was possible when the "Enable SD support" button was checked, but the functionality goes away when it is unchecked.

### Did the issue persist even in safe mode?

Yes, it did persist

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

Occurs in safe mode.

### Version of OctoPrint

OctoPrint 1.9.3

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Sovol SV06 2.0.0.A

### Browser and version of browser, operating system running browser

Brave v1.62.165

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [ ] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

Hopefully I did this right, first time.  :)
[octoprint-systeminfo-20240217122734.zip](https://github.com/OctoPrint/OctoPrint/files/14319669/octoprint-systeminfo-20240217122734.zip)
[10.0.0.58-1708191087103.log](https://github.com/OctoPrint/OctoPrint/files/14319682/10.0.0.58-1708191087103.log)

cp2004, 2024-02-17:
That'll be something I missed in #4633, so I've fixed it with #4953

foosel, 2024-04-16:
Just merged your PR after fixing the issue with it. I've also marked this as bugfix release relevant, so it might also end up in a future 1.10.x bugfix release after stable is out.

foosel, 2024-04-25:
Backported to 1.10.1.

foosel, 2024-05-14:
1.10.1 has just been released.

PR #4953 (2024-04-16): :bug: Fix file upload button properties when SD not enabled
Seemed like a lot of redundant code to just change one class name, so rather than duplicating I've condensed it.

#### What does this PR do and why is it necessary?

Fixes #4952 and makes future recurrence of the same type of issue unlikely.

#### How was it tested? How can it be tested by the reviewer?

It has not (yet) been tested, but should be fine. No drastic changes, but I will check it runs tomorrow.

#### Any background context you want to provide?

#4633 

#### What are the relevant tickets if any?

#4952
closed 2024-05-14 by foosel