OctoPrint/OctoPrint #4955: gcode analysis fails with IndexError exception on OrcaSlicer produced gcode

zerog2k, 2024-02-20:
### The problem

with OrcaSlicer (1.9.1) produced gcode, octoprint gcode analysis fails

```
2024-02-19 22:42:35,928 - octoprint.filemanager.analysis - INFO - Starting analysis of local:thermoflask spout seal_TPU_5m5s.gcode
2024-02-19 22:42:35,929 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /Users/jens/venv/octoprint/bin/python3.11 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.01 --throttle-lines=100 --bed-z=0.0 --offset 0.0 0.0 /Users/jens/Library/Application Support/OctoPrint/uploads/thermoflask spout seal_TPU_5m5s.gcode
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/Users/jens/venv/octoprint/lib/python3.11/site-packages/octoprint/__main__.py", line 6, in <module>
    octoprint.main()
  File "/Users/jens/venv/octoprint/lib/python3.11/site-packages/octoprint/__init__.py", line 1008, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "/Users/jens/venv/octoprint/lib/python3.11/site-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jens/venv/octoprint/lib/python3.11/site-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
         ^^^^^^^^^^^^^^^^
  File "/Users/jens/venv/octoprint/lib/python3.11/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jens/venv/octoprint/lib/python3.11/site-packages/octoprint/cli/common.py", line 37, in invoke
    return self._impl.invoke(ctx)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jens/venv/octoprint/lib/python3.11/site-packages/click/core.py", line 1688, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jens/venv/octoprint/lib/python3.11/site-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jens/venv/octoprint/lib/python3.11/site-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/jens/venv/octoprint/lib/python3.11/site-packages/octoprint/cli/analysis.py", line 129, in gcode_command
    interpreter.load(
  File "/Users/jens/venv/octoprint/lib/python3.11/site-packages/octoprint/util/gcodeInterpreter.py", line 317, in load
    self._load(
  File "/Users/jens/venv/octoprint/lib/python3.11/site-packages/octoprint/util/gcodeInterpreter.py", line 395, in _load
    filamentValue = comment.split("=", 1)[1].strip()
                    ~~~~~~~~~~~~~~~~~~~~~^^^
IndexError: list index out of range
2024-02-19 22:42:36,149 - octoprint.filemanager.analysis - ERROR - Analysis for local:thermoflask spout seal_TPU_5m5s.gcode ran into error: No analysis result found
```

[thermoflask spout seal_TPU_4m54s.gcode.zip](https://github.com/OctoPrint/OctoPrint/files/14339686/thermoflask.spout.seal_TPU_4m54s.gcode.zip)

Looks like the code at https://github.com/OctoPrint/OctoPrint/blob/master/src/octoprint/util/gcodeInterpreter.py#L391-L404
only expects Slic3r to have this, but gcode produced by OrcaSlicer has comments like:
```
 ; filament_diameter: 1.72
```

PR candidate incoming  ;)

### Did the issue persist even in safe mode?

Yes, it did persist

### If you could not test in safe mode, please state why ("currently printing" is NOT an excuse!)

exception in core octoprint code

### Version of OctoPrint

1.9.3

### Operating system running OctoPrint

macOS 11.7.10

### Printer model & used firmware incl. version

Powerspec Ultra3d

### Browser and version of browser, operating system running browser

Safari 16.6.1

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

zerog2k, 2024-02-20:
ok looks like this may already have been addressed w/
https://github.com/OctoPrint/OctoPrint/commit/f4fbe9193856e828e5c83444be0cab7acedcb50d

closed 2024-02-20 by zerog2k

RagingCub, 2024-11-27:
Any idea when this is going to be merged?

foosel, 2024-11-27:
The goal is to release 1.11.0rc1 as soon as possible, ideally still this year (but it's only one person working on this full time - oh hi, that's me - and my day has the same length as everyone else's). Sadly this report only reached me when 1.10.x was already frozen.