OctoPrint/OctoPrint #4957: Bump WebSocket Client To > 1.6.1

QuinnDamerell, 2024-02-23:
### The problem

I found an issue in the PY Websocket Client lib a while back that we were able to fix:
https://github.com/websocket-client/websocket-client/pull/918

But the fix is in 1.6.1 and above. I was looking at the OctoPrint code, and the lib is locked to 1.5.3, saying that anything higher requires PY 3.8.

I'm not sure what the minimum requirements are for OctoPrint, but if we could bump the package, it would benefit users since the bug in the WebSocket client can cause the WebSocket deadlock on shutdown.

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

1.9.3 and above

### Operating system running OctoPrint

OctoPi

### Printer model & used firmware incl. version

Prusa

### Browser and version of browser, operating system running browser

Chrome & Firefox

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [X] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [X] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

jneilliii, 2024-02-24:
Pretty sure the reasoning was because OctoPi 0.18 was Python 3.7.

QuinnDamerell, 2024-02-24:
The release page for it says 1.6.1 works with 3.7+, but I don’t know how to actually verify that. 

https://pypi.org/project/websocket-client/1.6.1/

jneilliii, 2024-02-24:
Yeah, looking at the 1.6.1 version it does appear to be 3.7 compatible based on the classifiers. Should be ok to do `>=1.6.1` (I see you updated the issue title on us 😉).

jneilliii, 2024-02-24:
This could theoretically be fixed in current RC phasing, because the version change on pinning was added just 3 weeks ago to maintenance/rc branches. 

https://github.com/OctoPrint/OctoPrint/commit/6ae8822154027eb0e89ae7ca23151069de0abedd

QuinnDamerell, 2024-02-24:
Hahaha, yeah, I correct the title so it was accurate. I will try to make a quick PR to change it

foosel, 2024-03-06:
Tbh, I have a bit of a stomachache about bumping this. The Python version change came officially with a patch, and I'm a tad worried that there are changes in 1.6.0 and 1.6.1 that already will cause issues with Python 3.7, as a Python version limit change on a patch version without an entry in the changelog feels VERY much like "ooops, no longer works with that, oh well, ok then, let's just change the minimal Python version and the classifiers then, no one will care".

In short, this kind of breakage made some alerts go off in my head when I prepped the first RC1 and I decided to stay clear of 1.6.x for now because of it.

If someone could check if everything looks fine with 1.6.1 under Python 3.7, that would help. I'm still unsure about it however.

Currently still recovering from my surgery last week as well, so sorry for the late reaction.

QuinnDamerell, 2024-03-11:
Yeah, that makes sense. I force 1.6.1 at minimum when you install OctoEverywhere, and I haven't heard of any issues with it. That's something, but more is needed to prove it works 100%. 

The way OE uses the WebSocket, the bug is hit all the time and would really jam up OctoPrint since the thread got stuck.

Beyond that, I don't have a great way of testing the 3.7 compatibility. Do you know of any other good ways to test?

foosel, 2024-03-12:
So here's what I've done now:

- Gone through the whole [comparison between 1.5.3 and 1.6.1]() without finding anything that looks like it won't work with Python 3.7
- Pulled 1.6.1 and run `pyupgrade --py37-plus` on it, it didn't change anything

So the syntax is fine at least. Out of curiousity I also went through the changes from 1.6.1 to 1.6.2 and the only thing stopping it to work with Python 3.7 I could find was limiting the supported Python version in `setup.py`. So this indeed looks like a case of "the code in principle still works with 3.7, but since that's official EOL we don't support this anymore". Ok.

Given that I only use it inside OctoPrint itself for the included `octoprint_client` package and that seems to work fine with 1.6.1, I'll bump this to 1.6.1 for OctoPrint 1.10.0 (would have merged you PR, but you created it against `devel` for some reason 😅). For the future though, we'll need to figure something out, because from 1.6.2 on this lib definitely no longer supports 3.7, and that's still in use by half of the current OctoPrint userbase. 

At least Python 2 usage is finally down to under 5% 🙄

QuinnDamerell, 2024-03-12:
Wonderful, thank you @foosel!! 

I have noticed that the new PY versions seem be to coming and going way more quickly. I read online that they are trying to EOL versions soon, so they don't get stuck in the big 2.0 issue. But it will be interesting to see how projects adapt, like I don't think OctoPrint really has way to update the PY version at the moment does it?

foosel, 2024-03-12:
I could always just update the version, as I now have tooling in place to quickly adapt to syntax changes and such, but that would lock out a HUGE part of the user base and so while technically possible, it's something I just can't do without causing a ton of harm out there.

It's one thing declaring a Python version EOL, it's another to declare a library as no longer supporting that version without using any language features only available in newer versions, and I found that decision frankly quite annoying. It only makes the job harder for maintainers like myself. But it's not my decision to make, just my job to cope with it somehow - via vendoring if push comes to shove.

QuinnDamerell, 2024-03-12:
Yeah, I totally understand that. It's annoying that the packages are dropping min support so quickly. Thanks for the info.

commit 8a9478c4 (2024-03-12): ⬆️ Bump websocket-client to 1.6.1

Closes #4957
closed 2024-03-12 by QuinnDamerell