OctoPrint/OctoPrint #4958: [Request] Ability to reload GCode viewer after cancel

Techineer, 2024-02-24:
### Is your feature request related to a problem? Please describe.

If I need to cancel a print, the GCode viewer won't display print progress after the print is started again.
In order to re-parse the gcode for the GCode viewer, either OctoPrint must be restarted, or the file must be deleted and reloaded.
I notice that once canceled, the print status information continues to appear above the files list and cannot be cleared to reset the print and reload.  

Is it possible to provide a way to completely restart the reload/print process and re-parse the file for the G-Code viewer after a cancel?

### Describe the solution you'd like

Following cancellation of a print, either:
Provide a 'reset' control to clear status information and reinitialize the print process or,
reinitialize the print process when starting the print again, including reloading and parsing the GCode viewer.

Techineer, 2024-02-24:
Additional info:
Running OctoPrint v1.9.3 in Firefox 123.0 under Windows 10
Directly connected via USB cable to Ender 3 S1.

cp2004, 2024-02-24:
Rather than a feature request, this sounds like a bug. All other ways of starting a new print work fine, and the gcode viewer picks up the new print.

Printing > Cancelling > Load new file > Print  - this works fine

Printing > Cancelling > Load & Print  - this doesn't work, the gcode viewer does not get the new file.

foosel, 2024-06-06:
I can no longer reproduce this, but I'm pretty sure it was the same as #4978 and has thus been fixed in 1.10.0

closed 2024-06-06 by foosel