OctoPrint/OctoPrint #4962: Serial Verbndung

raziel21183, 2024-02-26:
### The problem

Mein Octoprint unterbricht einfach die Serial verbindung zum Drucker

[octoprint (3).log](https://github.com/OctoPrint/OctoPrint/files/14409956/octoprint.3.log)
[octoprint (2).log](https://github.com/OctoPrint/OctoPrint/files/14409972/octoprint.2.log)

### Did the issue persist even in safe mode?

Yes, it did persist

### Version of OctoPrint

 OctoPrint 1.9.3 Python 3.9.2 OctoPi* 1.0.0 (build 2023.10.09.151442

### Operating system running OctoPrint

Python 3.9.2 OctoPi* 

### Printer model & used firmware incl. version

Anycubic i3 mega s

### Browser and version of browser, operating system running browser

Chome

### Checklist of files to include below

- [X] Systeminfo Bundle (always include!)
- [ ] Contents of the JavaScript browser console (always include in cases of issues with the user interface)
- [ ] Screenshots and/or videos showing the problem (always include in case of issues with the user interface)
- [X] GCODE file with which to reproduce (always include in case of issues with GCODE analysis or printing behaviour)

### Additional information & file uploads

[octoprint-systeminfo-20240226203718.zip](https://github.com/OctoPrint/OctoPrint/files/14410017/octoprint-systeminfo-20240226203718.zip)

raziel21183, 2024-02-26:
Es kommt bei mehreren Drucken vor das der Drucker einfach die Verbindung verliert. USB anschlusskabel wurde bereits getauscht und auch einen anderen USB Anschluss am Rasberry Pi ausprobiert.

foosel, 2024-03-06:
Ich sehe leider in keinem der Logs irgendeinen Hinweis auf einen Disconnect, oder Safe Mode, wie bereits angemerkt. Im Allgemeinen sind solche Fehler jedoch auch umgebungsspezifisch - die Verbindung bricht dann unter OctoPrint weg, auf OS Ebene, und damit ist das nichts, was von OctoPrint behoben werden kann.

Bitte starte in safe mode, reproduziere das Ganze, erzeuge ein neues System Info Bundle und setze dich dann AUF DEM COMMUNITY FORUM mit uns in Verbindung. Falls es irgendwie geht, bitte in Englisch, da das den Kreis derer, die dir helfen können, ganz *erheblich* vergrößert.

closed 2024-03-06 by foosel